artisan-roaster-scope/artisan #312: Alarms: having the option to activate an alarm only if both conditions are satisfied (temperature and time)

lucapinello, 2019-02-13:
Would it be possible to activate an alarm only if both conditions are satisfied (temperature and time)?

Thank you so much!

RickGroszkiewicz, 2019-02-23:
This would be useful to automate a "time based" melange roast on my Hottop. I am doing this manually for decaf coffee, and could not get the artisan alarms to automate the process.

For example, when BT hits a certain value, open the door to eject some beans. After 3 seconds, close the door. After 15 seconds, drop the remainder of the beans.

MAKOMO, 2019-02-25:
Well, the Artisan alarm system is complete w.r.t. propositional logic. Disjunction is build in, conjunction and full negation is there via alarm chains. Check my post "More Alarms" on alarm chains: https://artisan-roasterscope.blogspot.com/2016/08/more-alarms.html

@RickGroszkiewicz : the example you give is supported already (thus I do not know how long your Hottops door solenoid will take this. Configure a first alarm (1) to open the door at a certain BT. Configure a second alarm (2) that chains to the first positively by putting a 1 into the "If Alarm" field. Select "If Alarm" from the "From" popup for this alarm (2) to have the time criteria be taken relative to the chained alarm, thus by putting 0:03 into the Time field, this refers to the moment in time, 3sec after alarm (1) triggered. Let alarm (2) close the door again. Add a third alarm (3) with chained to the second by setting its "If Alarm" to 2, again select "If Alarm" from the "From" popup and put 0:15 as Time criteria. Make this 3rd alarm again open the door via its action.

closed 2019-02-25 by MAKOMO

MAKOMO, 2019-02-25:
@lucapinello just add a first alarm testing the your first condition without any action and chain a second alarm to it testing the second condition and setting the alarm action.

lucapinello, 2019-02-25:
Thanks a lot @MAKOMO, this is what I end up doing and it works reasonably well. I was trying to create a button to temporarily disable my alarms. However, since the alarms are executed only once, as soon as I trigger the first time this button, I cannot enable future alarms anymore. Not a big problem, but I was wondering if there is an option for this? Thanks a lot for developing this software, the more I use it the more I love it!

MAKOMO, 2019-02-25:
How to create a button to temporarily disable alarms? What would be the alarm action?

The alarm dialog contains a button to turn alarms "ALL ON" or "ALL OFF". Note that each alarm has a tick mark to activate or deactivate it.

Yes, alarms do trigger only once. This is by design.

lucapinello, 2019-02-25:
Thanks for the quick reply!

Let me explain the use case. I have created this little software to create
profiles: https://github.com/lucapinello/SR700_Artisan_Profile_Builder.

This together with the other server code allow artisan to control the
roaster precisely.

This software spits out an alarm file that controls temperature and fan
every 20 seconds,  and at the end automatically drop and stop the roaster.
This allows to have very precise profiles, see for example this one
attached inspired by Rao's book.

[image: image.png]

Now some people have asked me to implement a temporary manual mode during
the roasting process to fine tune the process on the fly. In the current
setup as soon as you disable the alarm they are all gone.

I was looking for an option to temporarily ignore some of the alarms that
are triggered by time when this "manual mode" is enabled and to resume the
alarms after that. Does it make sense?

Thanks again,

Luca

MAKOMO, 2019-02-25:
Luca, 

so you would like to have a silent button for alarms. Hm. So you would maybe like to just deactivate the activation of alarm actions temporary and keep triggering the alarms. Otherwise a bunch of alarms would potentially trigger later at higher temperatures when you turn alarms loud again. I just added an "Artisan Command" button action that understands the verb alarms(<bool>). So a button with action "Artisan Command" and documentation alarms(1) will turn alarms loud and alarms(false) or alarms(0) will turn alarms silent. Would that help? Binaries should be on BinTray in some minutes.

lucapinello, 2019-02-26:
> alarms(). So a button with action "Artisan Command" and documentation

Hi Marko,

Yes, I think this should work very well.

Thank you so much!

Luca