artisan-roaster-scope/artisan #314: Various bugs relating to negative PID duty

Snagi0, 2019-02-22:
## Expected Behavior
No settings conflicts or bugs.

## Actual Behavior
Negative PID target functionality brakes if input limits are set to positive values in config>curves>filters menu (PID duty is limited to the same min value). This also causes, lower than min limit, heater and fan values to disappear from the graphs (during roast, after they are drawn correctly). When input limits are disabled, PID output duty can go negative, but negative values are still not activated on target slider. If "invert control" is checked, negative duty functions normally but targets need to be also inverted so the control is correct. 

## Steps to Reproduce the Problem

  1. set positive min input limit
  2. setup pid with positive and negative targets (-100 to 100 pid duty limits)
  3. start roast (and charge) with suitable ramp and soak values to see how the pid functions

## Specifications

  - Artisan Version: 1.5.0
  - Artisan Build (number in brackets shown in the about box): fc6ebda9
  - Platform (Mac/Windows/Linux + OS version): windows 10
  - Connected devices or roasting machine: diy roaster connected with TC4 equivalent interface
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-02-25:
As in your other issues, your descriptions are partial and hard to understand and you seem to lack some basic understanding of the core Artisan concepts. The input filter has nothing to do with the PID functionality. The input filter cuts of values out of its range on sampling data from connected devices. The (software) PID is a control loop that tries to achieve a target source value (SV) by generating a duty signal, which in Artisan can be used to move sliders and potentially issue control commands to a connected device. If you limit the duty to positive values, a negative target slider will never receive a signal. By design. Thus there was an issue that prevented the negative target sliders to move at all. Fixed.

closed 2019-02-25 by MAKOMO

Snagi0, 2019-02-25:
I agree again, the input filter SHOULD NOT have nothing to do with the PID functionality.

Did you try to replicate this?

Your release history shows some PID related fixes with TC4 so I don't know if you have already done someting for this. v1.6.1: "fixes the p-i-d command for the TC4 that got broken in v1.5"

input filter limits: checked
input filter min: 10C
PID target positive: heater 
PID target negative: fan
invert control: not checked
PID duty steps: 1%
PID max steps: 100%
PID min steps:  -100%
>>>>>>>>>>>>PID duty output does not go under 10C

input filter limits: not checked
PID target positive: heater
PID target negative: fan
invert control: not checked
PID duty steps: 1%
PID max steps: 100%
PID min steps: -100%
>>>>>>>>>>>>PID duty output does to -100%, but fan does not turn on

input filter limits: not checked
PID target positive: fan
PID target negative: heater
invert control: checked
PID duty steps: 1%
PID max steps: 100%
PID min steps: -100%
>>>>>>>>>>>>everything works, but it makes no sense that PID targets need to be inverted

PS. Now I don't really like your attitude.

MAKOMO, 2019-02-25:
It is really hard to understand what you are saying as you keep on giving partial information. Just attaching your Artisan settings file could avoid quite some misunderstandings. 

First of all, the v1.6.1: "fixes the p-i-d command for the TC4 that got broken in v1.5" fix has nothing to do with this here, as it is, well, about the p-i-d command.

I needed a moment, again, to understand what you mean by "PID duty output does not go under 10C" in the first of the three experiment description. Now I understand. You are talking about your extra device of type "PID SV/DUTY %". Sure this signal is cut by the input filter as the signal of all other curves as I tried to tell you before. That does not mean that the PID is not emitting the correct duty signal internally. The "PID SV/DUTY %" signal is only an additional signal emitted by the PID mechanism to allow to track its behavior, but not used for anything else. If you decide to cut-off values below 10, also this signal gets cut-off.

As I told you, I fixed the negativ target slider with my last commit on closing this issue. As always you can verify this by running from source, wait for the next release, or check out the automatically generated (and potential unstable) pre-builds on BinTray: https://bintray.com/artisan/artisan-artifacts/artisan/master#files

Snagi0, 2019-02-26:
Yes I was talking about that signal. But you see, it really is not allowing to track PID behavior if its limited by those limit values and real PID output does not correspond to that value. 

I think I missed the bug fix comment when it was mixed in with all the insults. Better approach might be to ask for clarification. In this case I would have had to change back my settings and save multiple settings files so I did't do it. Thanks anyway.