artisan-roaster-scope/artisan #315: Various bugs relating to designer

Snagi0, 2019-02-22:
## Expected Behavior
In designer config window, I can click on temperature or time and erase the value using backspace before writing down the new value. 

It should be possible to set the dry end time lower than 10 minutes.

Also no error messages and weird looking graphs when all the time and temperature values are changed at the same time and pressed update.

## Actual Behavior
 If the cursor is located on the right of any number (temperature values), backspace does not do anything. If you erase the comma first, then you can erase the rest of the numbers. With time values it's not possible to erase the last number value from the minutes side, before erasing colon sign. 

If you press update with dry end time lower than 10 min, notification is shown. If you drag the point on the graph, there is no limitation like this.

I received an error message relating to designer config values when pressing update. Sometimes the graph curves also look weird and like if they are expecting to go to wrong direction (positive/negative slope)

## Steps to Reproduce the Problem

  1. Put the cursor left to the colon sign (between minutes and colon) and press backspace.
  2. Put the cursor left to the comma sign and press backspace. 
  3. Set dry end time to example 5:00 and press update.
  4. For the error I don't have exact values saved and it does not repeat always. However I think that it and the weird looking graphs are caused because I don't have the typical pre heating and bean temperatures catching up to the TP with my roaster. Instead my graphs start at near room temp and go up from there, I want to use designer the same way. Current interpolation method and limit values in designer does not seem to be general enough. 

## Specifications

  - Artisan Version: 1.5.0
  - Artisan Build (number in brackets shown in the about box): fc6ebda9
  - Platform (Mac/Windows/Linux + OS version): windows 10
  - Connected devices or roasting machine: diy roaster connected with TC4 equivalent interface
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-02-25:
I think you forgot to mention that you are talking about the Designer Config dialog here. I cannot reproduce any of those issues under Mac OS X. I will ask a colleague to look at this under Windows..

Snagi0, 2019-02-25:
The first sentence starts :"In designer config window". I'm talking about the settings popup menu, that can be accessed in designer by right clicking and pressing config. It has the title "Designer Config".

MAKOMO, 2019-02-25:
Sorry, now I am seeing it. My eyes are getting old.

roasterdave, 2019-03-03:
1. In version 1.5 either highlight the entire field then press delete or place the cursor to the right of the colon and use backspace.  The next version of Artisan will allow backspace to erase digits regardless where the cursor is placed.

2. I cannot reproduce this on Windows or RPi.

3. Two digits are expected to the left of the colon.  Use "05:00".  The next version of Artisan will allow a single digit to the left of the colon.

4. Works fine for me designing bean temp curves that start at room temperature (fluid bed).  You might want to add additional points to give you greater control over the shape of the curve when needed.

commit f613012a (2019-03-03): changes Designer Config field validation, resolves #315
closed 2019-03-03 by roasterdave