artisan-roaster-scope/artisan #316: Event handling bug/feature request

Snagi0, 2019-02-22:
## Expected Behavior
When quantifier (for example PID SV source) is set to slider with serial command action, serial output commands are sent every time the slider value is changed.

## Actual Behavior
The slider is updating  and receiving values correctly. However not a single serial command is sent out. I expect that this is because the slider is never "released", thus preventing slider action to function. 

I think that simple fix would be to just update and release the control of the slider every time value is changed. This would allow even more flexibility with control methods overall. 

There should be an option to output controlled heater (and fan) duty values without using PID or alarm/manual controls. I think the best way would be to give an option in ramp & soak to be able to choose between SV and duty. I tried to circumvent this by feeding ramp & soak SV (0-100C) value via slider to the controller and use that SV value as 0-100% duty, but run across this problem.  

If there is some other way to do this, please let me know. However I think that this "bug" would be good to fix.

## Steps to Reproduce the Problem
1. Setup PID SV extra device
2. Setup a quantifier with SV source
3. Setup slider with same event and serial output action
4. Start roast and confirm that the value updates but nothing is sent (for example with serial monitor)

## Specifications
* Artisan Version: 1.5.0
* Artisan Build (number in brackets shown in the about box): fc6ebda
* Platform (Mac/Windows/Linux + OS version): windows 10
* Connected devices or roasting machine: diy roaster connected with TC4 equivalent interface

Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-02-25:
Quantifiers set events based on the data of the observed curve. The slider move automatically to the latest event value. However, sliders moved by quantifiers do not emit action values by design. This is to avoid loops in the intended application scenario where one uses Artisan sliders and quantifiers to work in par with a machines manual controls. This won't change. Sorry. The quantifier mechanism has nothing to do with the control mechanisms you are talking about further down like PID or R/S.

It does also make absolutely no sense to have the R/S mechanism to output duty values instead of SVs. The R/S mechanism is an integral part of the PID loop that get's SV as input and produces duty values as output. All this sounds very confused to me.

If you are looking for a way to have emit Artisan control values you can either use an alarm program or the reply events feature assuming you have corresponding actions attached to the corresponding event sliders.

If you need anything else you might need to implement your own software.

closed 2019-02-25 by MAKOMO

Snagi0, 2019-02-25:
Okay, the quantifiers behavior is understandable. I agree, it has nothing to do with control mechanisms, I was only trying to use it to output those duty values.

To be perfectly clear, I am not purposing that you change the current PID R/S mechanism at all. 

What I wanted to do is, NOT use the PID control at all and control the heater directly by outputting duty values along some predefined curve. This can be achieved adequately using similar ramp control as you are using with PID. You must agree that setting up hundred of alarms to achieve this would be quite a bit more tedious. Also because of the fact that all the alarm actions are single time triggered, I think. I agree that this kind of control is less advanced than PID control, but does not need PID tuning and when the environmental variables are kept constant this type of control also achieves repeatable results. I know for a fact that other roasting softwares allow this kind of control AS AN OPTION. And saying that control like this makes absolutely no sense is the same thing than to say that you should not be able to use sliders or manual controls on the roaster to control the heat output by hand.

If I am missing something or do not understand correctly what you mean by "reply events" please correct me.

No hard feelings, I just think it's a shame if you want to take a harsh stance on this and disallow this type of open loop control completely. I just modified my roaster firmware to support Artisan and otherwise it seems quite nice. I'm also going to use the PID control but would like to have my options open.

MAKOMO, 2019-02-25:
Wait a second. Why not using events which are exactly designed for this. Construct a profile, can be an artificial one eg. using the designer. Now you add some events as you need to it using the Events table under the Roast Properties dialog. Now you load this one into the background while roasting and activate the playback events feature (either by time or temperature). If the events you were setting in that background profile have some slider definitions with actions associated, they will be fired at the corresponding times and temperatures during the roast. Am I still missing something?

Snagi0, 2019-02-26:
I guess that could be little better than alarms, but I think that you still need to enter all control values manually. I have to look into that.