artisan-roaster-scope/artisan #317: Reading more than two values from

lucapinello, 2019-02-25:
I am using with success the External Program function to get the values from ET and BT for a roaster that is not supported by Artisan (SR700): https://github.com/lucapinello/SR700_Artisan_Server.

However, the device has other readings I would like to map as virtual devices.

How can I read these additional measuraments? Is it possible to extend the logic of the External Program to have more comma separated values?

Something like this:

#ET,BT,AIR,AMBIENT_TEMP

180,150,10,100

Thanks a lot for considering this!

Luca

MAKOMO, 2019-02-25:
Did you try to add an extra device of type "Program 34", "Program 56",..?

closed 2019-02-25 by MAKOMO

pinellolab-service, 2019-02-25:
Terrific! This is exactly what I was looking for. Thanks again.