artisan-roaster-scope/artisan #319: Coffee Properties: do not remain filled when starting new batch

InvisibleOrdinary, 2019-02-27:
## Expected Behavior

When running through a sequence of batches, all Roast Properties that were filled in for the previous batch remain filled with same data (with the exception the ambient humidity, temp, & pressure data, which should update at the start of the new roast)

## Actual Behavior

When running through a sequence of batches, the roast properties are cleared for each new roast batch

## Steps to Reproduce the Problem

 0. Fill in roast properties
  1. roast
  2. cmd-N to begin next batch
  3. check cmd-t

## Specifications

  - Artisan Version: Artisan 1.6.0 (435ba27) 
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14.3
  - Connected devices or roasting machine: phidgets 1048 or VINT

[190227_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2912045/190227_artisan-settings.aset.zip)

MAKOMO, 2019-02-28:
Well, you have "Delete roast properties on RESET" ticked in your settings. Check the last line of the Roast Properties dialog, first tab.

closed 2019-02-28 by MAKOMO

InvisibleOrdinary, 2019-02-28:
oh, oops. thanks