artisan-roaster-scope/artisan #321: <carriage return> in dialogue boxes should accept & close dialogue (trigger 'ok')

InvisibleOrdinary, 2019-02-27:
## Expected Behavior

when filling in data in a dialogue box, typing the <carriage return> accepts the changes and closes the dialogue - this could apply to both <CR> as well as <Enter> or just <CR>

## Actual Behavior

nothing happens when in a dialogue box and type carriage return

## Steps to Reproduce the Problem

  1.  Open a dialogue box such as cmd-b
  2. type <carriage return>
  3. …

## Specifications

  - Artisan Version: 1.6.0 (435ba27) 

  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14.3
  - Connected devices or roasting machine: Phidgets 1048 or VINT with various devices
  
[190227_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2912094/190227_artisan-settings.aset.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-02-28:
added default focus to OK button and assignded CMD-W and CMD-. shortcuts to the Cancel button of most dialogs

closed 2019-02-28 by MAKOMO

InvisibleOrdinary, 2019-02-28:
Thanks

also <escape> is standard trigger for cancel (at least on Mac). maybe add that as well?

MAKOMO, 2019-02-28:
Sorry? On Mac OS X, CMD-. is a standard to close dialogs since, well, since before Mac OS X;) What is now still missing?

InvisibleOrdinary, 2019-02-28:
aah… I see that I misread your CMD… I thought you were writing CMD- (hyphen) but you wrote CMD-. (period). Yes, that works… but what I said above is still true that across all the apps I use, escape is more consistent than CMD-. I was just suggesting adding that as well. 

CMD-W is usually for closing a window (as opposed to a dialogue box)… 

Thanks

Dave

MAKOMO, 2019-02-28:
Yes, I am aware that CMD-W is actually for closing windows, but I added it anyway here too. Cannot hurt. Now I got it. ESC is missing. Yes, I will add ESC as well.

InvisibleOrdinary, 2019-02-28:
Cool. Thanks