artisan-roaster-scope/artisan #322: Feature Request: batch edit roast properties or past roasts

InvisibleOrdinary, 2019-02-27:
So when roasting several batches sequentially, it's not always easy or possible to enter post-roast data into the roast properties fields esp roasted weight while that roast is still open… it would save quite a bit of time, and make workflow much more efficent, to be able to open a dialogue that would give access to at least the following roast properties (below) with the ability to edit them for multiple roasts at a time. This would bypass the need to open each roast .alog and edit individually. It would also give the ability to correct any errors initially made

**ROAST PROPERTIES - General**

- Date
- Batch
- Title
- Beans
- Operator
- Machine
- Drum Speed
- Weight <unit> <in> <out>
- Volume <unit> <in> <out>
- Density  <unit> <val> <unit> <val>
- Whole Color
- Ground Color
- Moisture <green> <roasted>
- Ambient Conditions <%> <C/F> <kPa>
- Screen
- Bean Temp

**ROAST PROPERTIES - Notes**

- Roasting Notes / all check boxes
- Cupping Notes

Seems that Events & Data is more detail than is necessary in the workflow I am trying to streamline… though could see them as being desired by some as well…

  - Artisan Version: 1.6.0 (435ba27)
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14.3
  - Connected devices or roasting machine: Proaster + Phidgets 1048 / Pinalense + VINT and various

MAKOMO, 2019-02-28:
Use a second version of Artisan (just make a copy of Artisan.app and name it ArtisanViewer.app). If you start that second Artisan version while the first version is still running, it starts up in ArtisanViewer mode (introduced in v1.5). You can use that second instance to load, edit and save all Artisan profiles while roasting with the other instance.

closed 2019-02-28 by MAKOMO

poundy, 2019-02-28:
Marko, it might be worth including this as a standard part of the install package, so people don't get left with an orphaned older version of the app that gets forgotten?

MAKOMO, 2019-02-28:
Well, OS X does not come with an installer. Just drag-and-drop. On Windows you do not need to install two versions as Windows allows to start a second version of an app (Mac OS X not).

InvisibleOrdinary, 2019-02-28:
Hi Marko,

I will try that as a workaround, and maybe it will work better than I imagine, but I'm not seeing that as a long-term solution. I had hoped you would take the request in more serious consideration for future development of a tool within Artisan… 

The problem with this approach is that you still need to individually open/navigate into the Roast Properties/edit/save/open next… to edit the data. And that may work when you are doing a couple batches, or if everything that you will enter is easily accessible (weighing equipment, roasted beans, etc…), so having the ability to more easily edit post-roasting session would be of great help in production.

Maybe there is another way to access the data?

MAKOMO, 2019-02-28:
What you are asking for really is being developed at the very moment as an add-on to Artisan for commercial users. Ping me if you have an interest to test this.