artisan-roaster-scope/artisan #325: menu bar disappeared after recent OS upgrade

marcov, 2019-03-07:
I recently upgraded my OS (openSUSE Tumbleweed), and since then artisan menu bar has disappeared. No other programs misbehave.

This happens both for artisan 1.5.0 and the master version (started with `python3 artisan.py`).

Is there something I can do to debug this?

[image]

MAKOMO, 2019-03-07:
That must be something OS specific. Artisan is configuring the menu bar via PyQt and Qt. So if something goes wrong it is most likely in the hands of Qt. Not much we can do on the Artisan side I fear. So I close this here.

closed 2019-03-07 by MAKOMO

marcov, 2019-03-08:
I actually found a way to get the menu bar back.
```
diff --git a/src/artisanlib/main.py b/src/artisanlib/main.py
index 1b83a458..20412c5d 100644
--- a/src/artisanlib/main.py
+++ b/src/artisanlib/main.py
@@ -11562,6 +11562,7 @@ class ApplicationWindow(QMainWindow):
         self.ToolkitMenu = self.menuBar().addMenu(UIconst.TOOLKIT_MENU)
         self.viewMenu = self.menuBar().addMenu(UIconst.VIEW_MENU)
         self.helpMenu = self.menuBar().addMenu(UIconst.HELP_MENU)
+        self.menuBar().setNativeMenuBar(False)
         
         #FILE menu
         self.newRoastMenu = self.fileMenu.addMenu(UIconst.FILE_MENU_NEW)
```

> That must be something OS specific. Artisan is configuring the menu bar via PyQt and Qt. So if something goes wrong it is most likely in the hands of Qt.

So it may actually be some issue with recent version of PyQt or Qt.

 Not much we can do on the Artisan side I fear. So I close this here.

MAKOMO, 2019-03-08:
Dear Marco,

So with that line "setNativeMenuBar(False)" the menu bar appears again. But you get those ugly Qt menu bars, right?

I could add that line just for Linux builds, but then all Linux users will get those Qt menu bars although the native ones work fine on Ubuntu for example. Hm.

So it may actually be some issue with recent version of PyQt or Qt.

I recently updated to Qt 5.12, but that one broke something with the Linux builds and made all popups cutting part of their item texts. So I had to downgrade the builds to 5.11. Found Qt issues for both issues, but nobody seems to work on those. Hm.

Which version of Qt works for you without that NativeMenuBar False?

Not much we can do on the Artisan side I fear. So I close this here.

Anyhow, thanks for looking into this.

marcov, 2019-03-08:
> So with that line "setNativeMenuBar(False)" the menu bar appears again. But you get those ugly Qt menu bars, right?

No, to me it looks clean. Both other Linux users may have a different behaviour.

[image]

> Which version of Qt works for you without that NativeMenuBar False?

I haven't done a bisect of what version worked, it's a bit annoying to pin a specific library when running on a "rolling" distribution.

> Not much we can do on the Artisan side I fear. So I close this here.

It's ok, thanks for answering. Running off the git tree with this patch it's ok for me.