artisan-roaster-scope/artisan #326: Events... is not working on 1.6.1

lucapinello, 2019-03-10:
When I click on config Events... one error window is displayed

This is the log file:
2019-03-10, 01:32:56
--------------------------------------------------------------------------------
<class 'AttributeError'>: 
type object 'QKeySequence' has no attribute 'Cancel'  File "/home/pine64/ROASTER_SOFTWARE_NEW/artisan/src/artisanlib/main.py", line 25158, in eventsconf
    dialog = EventsDlg(self)
  File "/home/pine64/ROASTER_SOFTWARE_NEW/artisan/src/artisanlib/main.py", line 33216, in __init__
    cancelAction.setShortcut(QKeySequence.Cancel)

--------------------------------------------------------------------------------
/home/pine64/ROASTER_SOFTWARE_NEW/artisan/src/artisanlib/main.py::eventsconf:25158
	                self = <artisanlib.main.ApplicationWindow object at 0x7fb5d079d8>
/home/pine64/ROASTER_SOFTWARE_NEW/artisan/src/artisanlib/main.py::__init__:33216
	           titlefont = <PyQt5.QtGui.QFont object at 0x7fad6874a8>
	           __class__ = <class 'artisanlib.main.EventsDlg'>
	       barstylelabel = <PyQt5.QtWidgets.QLabel object at 0x7fad4f9ca8>
	           barstyles = ['', 'Flag', 'Bar', 'Step', 'Step+', 'Combo']
	                self = <artisanlib.main.EventsDlg object at 0x7fb1c87318>
	              parent = <artisanlib.main.ApplicationWindow object at 0x7fb5d079d8>
	     valuealphalabel = <PyQt5.QtWidgets.QLabel object at 0x7fac411dc8>
	    valuesymbollabel = <PyQt5.QtWidgets.QLabel object at 0x7fac411ca8>
	          typelabel1 = <PyQt5.QtWidgets.QLabel object at 0x7fad4f9dc8>
	          typelabel3 = <PyQt5.QtWidgets.QLabel object at 0x7fad4f9ee8>
	       defaultButton = <PyQt5.QtWidgets.QPushButton object at 0x7fac4129d8>
	     valuecolorlabel = <PyQt5.QtWidgets.QLabel object at 0x7fac411b88>
	            settings = <PyQt5.QtCore.QSettings object at 0x7fad4f9828>
	        cancelAction = <PyQt5.QtWidgets.QAction object at 0x7fac412a68>
	          typelabel4 = <PyQt5.QtWidgets.QLabel object at 0x7fad4f9f78>
	 valuetextcolorlabel = <PyQt5.QtWidgets.QLabel object at 0x7fac411c18>
	      valuesizelabel = <PyQt5.QtWidgets.QLabel object at 0x7fac411e58>
	         closeButton = <PyQt5.QtWidgets.QPushButton object at 0x7fac412948>
	 valuethicknesslabel = <PyQt5.QtWidgets.QLabel object at 0x7fac411d38>
	          typelabel2 = <PyQt5.QtWidgets.QLabel object at 0x7fad4f9e58>
Version: 1.6.1, revision: 0

This is with the default settings (I used the factory reset button to be sure this is not related to my settings)

lucapinello, 2019-03-10:
Up to this commit is working fine

git checkout 41393199aedcefbb4a4665174ebf78f707c5088d

MAKOMO, 2019-03-10:
Luca, which OS and Qt/PyQt version you are observing this?

lucapinello, 2019-03-10:
Thanks for the quick follow up!

I am using an Ubuntu based distro on an ARM CPU (Pine64). The version of the OS is:
Ubuntu 16.04.2 LTS (GNU/Linux 3.10.105-bsp-1.2-ayufan-77 aarch64)

>>> from PyQt5.QtCore import QT_VERSION_STR
>>> print(QT_VERSION_STR)
5.5.1

Please let me know if you need additional info.

Thanks,

Luca

MAKOMO, 2019-03-11:
That constant seems not to be defined on older Qt5 versions. Artisan is now on Qt/PyQt 5.11 and Qt/PyQt 5.12 is out for a while. Please update your Qt/PyQt stack. Nevertheless I added a try-catch for this.

closed 2019-03-11 by MAKOMO

lucapinello, 2019-03-11:
Thanks Marko!

I will update my PyQT.

Best,

Luca