artisan-roaster-scope/artisan #327: Inconsistent reading with phidgets voltage devices

InvisibleOrdinary, 2019-03-10:
I have a Phidgets slider (linear potentiometer 75mm) with symbolic formula to translate the voltage into a % of the distance along the length (x / 50, also tried other variations). This is being shown in an LCD so values should end up being 0-100 

The device shows correctly as long as the position is reading above 50% voltage. When the slider is pulled out so that the voltage drops below 50%, the value shown in artisan is frozen as whatever value last read above 50% - it's slow to respond so that number varies - and usually (but not always) errors to -1

I have tested the device several time with Phidgets dashboard, and the slider registers consistently well, so I am confident there isn't anything wrong with the device or the connection.

## Specifications

  - Artisan Version: Artisan 1.6.1 (ddd1677)
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14.3
  - Connected devices or roasting machine: phidgets: TMP1101, VINT, HUM000, ±7kPa pressure sensor
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

[Deft-Pinhalense-Sample_WIP.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2949214/Deft-Pinhalense-Sample_WIP.aset.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-03-10:
Could you uncheck "Limits" under "Input Filter" in the curves dialog, which limits all incoming readings (including those from the voltage input ports) to 0-1000 in your case and try again?

[image]

InvisibleOrdinary, 2019-03-10:
> Marko Luther <mailto:notifications@github.com>
> March 10, 2019 at 12:52 AM
> Screen Shot 2019-03-10 at 09 50 07

Oh ah… that is it. not sure why or when i changed the min to 50 -?

Thanks

Dave

closed 2019-03-10 by MAKOMO