artisan-roaster-scope/artisan #328: allow negative values to be displayed for all inputs

InvisibleOrdinary, 2019-03-10:
## Expected Behavior

Based on the existence of issue #221, I am assuming that negative values for other inputs are disallowed… I don't know what the reason for this would be, thought I realize that you wouldn't have a negative voltage, so for the input, that makes sense. But with, say differential pressure sensors, you would normally be able to show a positive or negative pressure, so the symbolic formula might return a negative (say, a negative pressure in your exhaust stack).

It would be good to be able to show this. 

## Actual Behavior

LED shows -1 for any value <0 that is calculated from an inpput

## Steps to Reproduce the Problem

  - Artisan Version: Artisan 1.6.1 (ddd1677) 
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14.3
  - Connected devices or roasting machine: phidgets TMP1101 x4, VINT, Linear Potentiometer. differiental pressure sensor…
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

[Deft-Pinhalense-Sample_WIP.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2949609/Deft-Pinhalense-Sample_WIP.aset.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-03-10:
Issue  #221 was resolved and was related to a GUI input field. Your issue relates to readings gathered from a device and potentially recomputed by symbolic formula. Symbolic formulas can generate negative values and those are also recorded and shown in the LCD. Just do a factory reset, set the main device to meter=Dummy and put "-20" and "-30" into the main channels symbolic formula fields. Those values show and record as expected for me.

[image]

closed 2019-03-10 by MAKOMO

InvisibleOrdinary, 2019-03-10:
hold up on closing. something is happening… perhaps I'm not testing this correctly, but I still get -1 when I set the main devices to DUMMY, and their symbolics to -30, -60 or whatever. 

I also get a port config dialog when I set it to DUMMY, which I have no clue how to configure (so said 'ok' to the defaults).

I also tried just overwriting the TMP 1101 main device with negative numbers, and that errored to -1 as well. but setting them to positive numbers showed correctly - which also worked for DUMMY.

InvisibleOrdinary, 2019-03-10:
oh, and also, how do you factory reset?

MAKOMO, 2019-03-10:
Factory reset is within the Help menu. The serial config dialog is shown for device Dummy as this one allows to send serial commands via actions that way.

MAKOMO, 2019-03-10:
In your attached settings file you still have the Input Filter Limit set. Untick that one and you see the negative values.