artisan-roaster-scope/artisan #330: Feature Request - Choose which curve to be presented on top

arkenamun, 2019-03-11:
BT curve is presented on top and may obscure other curves. Maybe there could be an option to choose which curve to present on top?

## Steps to Reproduce the Problem

  1. Multiple curves used in Artisan
  2. BT is presented on top
  3. BT may obscure valuable data from other curves eg ET2 in my example

The attached pdf illustrates how my thin ET probe provides useful data at the end of a roast in showing that ET has stopped declining, yet this can be hidden from view by the BT curve, resulting in this case in a delay in changing gas from G90 to G6 and a resultant small flick in BT.

## Specifications

  - Artisan Version: 1.6.1 (and previous releases)
  - Artisan Build (number in brackets shown in the about box): c23da469
  - Platform (Mac/Windows/Linux + OS version): Win10
  - Connected devices or roasting machine: Modbus and Phidgets
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[h2433_19-03-11_1210.pdf.txt](https://github.com/artisan-roaster-scope/artisan/files/2951709/h2433_19-03-11_1210.pdf.txt)

MAKOMO, 2019-03-11:
Hm. With any stack order, there will always be a curve rendered on top of another one. Most say, BT is the most important one. Thus the current order is BT, ET, deltaBT, deltaET, all others.

Why not just zoom in at that moment until you see all the details?

arkenamun, 2019-03-11:
Yes I understand those work arounds Marko and they are pretty good. I am proposing something that I think is more ideal. Zoom is great, but I then can't also see my gas pressure against the background data, which is super important for me (as I don't have closed-loop gas control so from roast to roast burner pressure varies).

Since Scott Rao proposed using ET and delta ET to pre-empt flicks in BT at the end of a roast (see links below), we have been using that method and roast results are improved. So ET and deltaET always provides the warning when BT is going to flick in advance. By the time BT has shown the flick it happened to the beans already, wasn't controlled, and will be tasted in the flavour. At that point BT change is tiny, almost finished, not so useful. This is illustrated in the following two articles. Look , I'm not intentionally comparing roasting software, however an alternative roasting software is used for these examples by Rao, I would assume to his financial benefit. I hadn't noticed previously the presentation of ET seems to be on top of ET in some of those graphs. Otherwise, in my opinion, the roasting software images are not as pretty or detailed (or affordable or flexible) as Artisan..... I love Dan Coplan's comment in the 2nd article, he just beat me to it!

https://www.cropster.com/news-detail/the-flick/
https://www.scottrao.com/blog/2018/11/25/how-to-use-cropster-to-know-exactly-when-first-crack-began

MAKOMO, 2019-05-09:
Curve order is not yet fully user configurable. However, curves are now drawn in LCD order thus ET is drawn above BT by default. If swap LCDs is ticked under Config >> Devices, the BT LCD is on top and thus the BT curve is drawn on top of the ET curve. Committed as 8b002aae825e43257127afb90e60640e22023954.

closed 2019-05-09 by MAKOMO