artisan-roaster-scope/artisan #332: Aillio R1 IBTS: not found or no permission

kirkjhayes, 2019-03-11:
## Expected Behavior
Installed Artisan on Windows 10 64B, configured both machine and meter for Allio R1 IBTS, expected Artisan to communicate with Allio R1

## Actual Behavior
Aillio R1: not found or no permission

## Steps to Reproduce the Problem

  1. Select Allio R1 IBTS
  2. Manually start Preheat on Bullet R1
  3. Select ON in Artisan
  4. Error on screen Aillio R1: not found or no permission

## Specifications

  - Artisan Version: 1.6.1
  - Artisan Build (number in brackets shown in the about box): c4dfe272
  - Platform (Mac/Windows/Linux + OS version): Windows 10 64B
  - Connected devices or roasting machine: Bullet R1 IBTS only
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/2954176/artisan-settings.txt)  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

rpaulo, 2019-03-12:
Does it work if you install the legacy roasttime USB drivers?

kirkjhayes, 2019-03-12:
> Does it work if you install the legacy roasttime USB drivers?

I do not know how to install legacy USB drivers with an IBTS setup. I assume legacy drivers will render the IBTS useless. Happy to try if someone would like to provide clarification on exactly which drivers to install that will still work with an IBTS.

rpaulo, 2019-03-12:
You can download it from here: https://aillio.com/?page_id=18556

It should not render the IBTS useless.

kirkjhayes, 2019-03-12:
Thanks, I'll try that.  I think this means I will have to use R1 and R2 software?

kirkjhayes, 2019-03-12:
I have installed legacy drivers "libusb-win32-bin-1.2.6.0" and this has fixed the communication issue!

ibrubeer, 2019-03-12:
You can confirm that Artisan works correctly with IBTS and the legacy
drivers then? Excellent! That's what I was waiting for.

kirkjhayes, 2019-03-12:
> You can confirm that Artisan works correctly with IBTS and the legacy drivers then? Excellent! That's what I was waiting for.
> […](#)

So far so good.
The Sliders for Drum Speed and Fan Speed might need a little tweaking as they go to 10 in the software, but I typically don't rely on the software when running manually anyway.

closed 2019-03-12 by kirkjhayes

HyperSprite, 2021-01-07:
If anyone is finding this in 2021 and trying to run Artisan with the Bullet R1, the link above for the legacy driver no longer works. I found the best way to install the drivers via the [Aillio RoastTime 3 install page](https://roastime.aillio.com/docs/setup) where it says **If they do not, please download [Zadig](https://zadig.akeo.ie/) and update the drivers.**. Turns out this tool will also install the legacy drivers, they just aren't the default selected.

I did this with the Bullet R1 V2 pugged in.

- Download and run [zadig](https://zadig.akeo.ie/)
- You should see the `Aillio LTD - Bullet R 1 ROASTER FS` selected
- On the left side you will see your current driver
- On the right side pick `libusb-win32 (v1.2.6.0)
- Install the driver