artisan-roaster-scope/artisan #337: Can't connect Artisan 1.6.1 to Aillio Bullet R1 v1.5 (IBTS)

josharmour, 2019-03-19:
## Expected Behavior
I should be able to select the machine in config > machine and then it is detected.

## Actual Behavior
'Aillio R1: not found or no permission' shows below menu above graph.

## Steps to Reproduce the Problem

  1. Get Bullet setup, working with roastime
  2. Download v1.6.1 of Artisan Scope
  3. Click Config > Machine > Aillio Bullet R1 IBTS
  4. Turn on Machine, try to connect.

## Specifications

  - Artisan Version: 1.6.1
  - Artisan Build (number in brackets shown in the about box): c23da469
  - Platform (Mac/Windows/Linux + OS version): Windows
  - Connected devices or roasting machine: Aillio Bullet R1 (IBTS)
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-03-19:
Did you successfully install the Aillio USB legacy driver as documented here?

https://artisan-scope.org/machines/aillio/

josharmour, 2019-03-19:
I did not.. Thanks!

josharmour, 2019-03-19:
The interesting thing is that the default software (RoasTime) for the
Bullet doesn't require an updated driver like this.  Because I have v1.5 of
the roaster, I'm told to use RoasTime 2.2.0.  I am willing to go through
this, but the driver is already working with RoasTime, so I worry that this
may mess with that.

josharmour, 2019-03-19:
Currently, in device manager, the Aillio shows up in Universal Serial Bus
Devices section as a recognized device with driver libwdi version
6.1.7600.16385..

So I should update that driver to the extracted legacy version?

josharmour, 2019-03-19:
Sorry for rapid fire updates :(

Device Manager will not let me update the driver with the legacy drivers,
it says I've already got an up to date driver..  I wonder if Aillio is
causing 3rd party roasting software to not work with the new driver and the
new software (RoasTime 2.0 and beyond)..  Can you update Artisan to use the
new driver?

josharmour, 2019-03-19:
https://imgur.com/a/jZtQNEr

josharmour, 2019-03-19:
And, after reading more, I see that I needed to run the inf-wizard.exe as admin (documented just fine) and it would give an option to install the driver at the end of the flow.  Closing now.

closed 2019-03-19 by josharmour