artisan-roaster-scope/artisan #338: Artisan v1.6.1 doesn't recognize Phidget 1048

drgarys, 2019-03-19:
## Expected Behavior Ability to read Phidget 1048

## Actual Behavior Doesn't read temperatures and displays an error message (see image below). Artisan-Mac 1.5.0 is able to configure and use the Phidgets with the same OS
[image]

## Steps to Reproduce the Problem

  1. launch app
  2. plug in Phidget to MacBook
  3. Start the roaster scope

## Specifications

  - Artisan Version: 1.6.1
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): 10.14.3
  - Connected devices or roasting machine: Phidget 1048
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-03-19:
Can you confirm that Artisan 1.5 connects successfully to the connected Phidgets on that very same machine where v1.6.1 fails as described above?

drgarys, 2019-03-19:
Yes, Marko. Same MacBook Air (late 2015), same OS.

MAKOMO, 2019-03-20:
I can duplicate this now. It is a codesigning issue. The security infrastructure of OS X 10.14 does not allow the current app to communicate with any device at all. I am working on this and will have to prepare another release. Thanks for reporting!

drgarys, 2019-03-20:
Thank you, Marko!

MAKOMO, 2019-03-20:
After spending most of the day on this, I found a way to get a signed version communicating with the Phidgets again that works on the current OS X 10.14. We will see how Apple is further developing this security framework and might at one point just have to give up this platform:(

I will prepare a new release v1.6.2 with the modified signing.

closed 2019-03-20 by MAKOMO

drgarys, 2019-03-20:
Hi Marko:

Do you have someone on your team who has Mac development expertise? I doubt Apple wants to totally disable apps like Artisan and they must have a huge community of people in the know.

(At worst I would resurrect my old MacBook just for Artisan and run keep it off the net for security.)

Thanks,
Gary

MAKOMO, 2019-03-20:
Gary, well, the Mac expert in the team is me. The problem is that Artisan is not developed using an Apple supported language nor its developer tools. Apple is concerned in stability and security of their system which is good. However, this leads to more and more restrictions on what an app can do and how it has to be constructed. Easier to handle for simple apps. Artisan requires quite some advanced capabilities and features from the OS.

Anyhow, I just released Artisan v1.6.2 that should fix the Phidget issue on OS X 10.14.

drgarys, 2019-03-20:
Marko,

Thanks again for your great efforts. You are a MVP and then some. Your
software is certainly eliminating any excuses for baking beans!

Gary