artisan-roaster-scope/artisan #339: Opctical Character Recognition (OCR) / Seven Segment Display reading through Camera

DeastinY, 2019-03-22:
Has this been discussed somewhere?  
This would allow to add many roasters in a non-invasive manner that were not originally designed to work with artisan.

tanius, 2019-04-09:
That's a really interesting idea! For inspiration, such a feature exists in the Android app [Roasted and Logged](https://play.google.com/store/apps/details?id=com.truebrew.ocr). Sadly, that one has not been updated since 2014.

DeastinY, 2019-04-09:
Thanks for picking it up! Roasted and Logged sadly seems not to be Open Source :o  
I've looked into reading seven segment reading algorithms and roughly plotted out a way to do it, *but* it involves quite a bit of homework. Probably nothing I could do as a sidegig. But if anyone was up for it, I'd gladly drop some insights.  
In regards to cameras - one could go the way of http://roastlogger.co.uk and decouple the camera input from the program. Roastlogger reads a part of the screen and does OCR on there, so it could be any webcam stream, a recorded video or USB - leaving ultimate freedom to the user

Koumi460, 2023-07-21:
Just randomly came across this thread years later... I'm the developer of Roasted and Logged app. I would have no problem making it open source, but the reason why I have not updated it since 2014 was because I was unhappy with the OCR. It was very primitive and most of the time it didn't work reliably. I tried some other approaches, but never got it to a state where I would be happy with how it works. So I'm afraid my code would be not much help...