artisan-roaster-scope/artisan #340: Error when starting designer

simenmm, 2019-03-23:
For the last half hour, whenever I start designer I get the following error. It started after I tried to start designer while having open measurements without events.

<class 'IndexError'>:
list index out of range

  File "artisanlib\main.py", line 25193, in designerTriggered
  File "artisanlib\main.py", line 25196, in startdesigner
  File "artisanlib\main.py", line 9127, in designer
  File "artisanlib\main.py", line 9229, in initfromprofile

--------------------------------------------------------------------------------
artisanlib\main.py::designerTriggered:25193
	                self = <artisanlib.main.ApplicationWindow object at 0x000001D7F724B4C8>
artisanlib\main.py::startdesigner:25196
	                self = <artisanlib.main.ApplicationWindow object at 0x000001D7F724B4C8>
artisanlib\main.py::designer:9127
	               reply = 16384
	                self = <artisanlib.main.tgraphcanvas object at 0x000001D7F743E798>
artisanlib\main.py::initfromprofile:9229
	                   i = 0
	                self = <artisanlib.main.tgraphcanvas object at 0x000001D7F743E798>

## Specifications

  - Artisan Version: 1.6.2
  - Artisan Build (number in brackets shown in the about box): 9d1202a4
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: CR3
  
[bug files.zip](https://github.com/artisan-roaster-scope/artisan/files/2999731/bug.files.zip)

MAKOMO, 2019-03-24:
I cannot successfully load the file "6th longer maillard.alog" attached in either v1.5.0 nor v1.6.2 of Artisan. Somehow this profile is broken. How did you generate this. Does this shows anything on loading in your Artisan v1.6.2 (or v1.5.0)?

simenmm, 2019-03-24:
That's strange, I can open it even on another computer.  I generated it by
loading a set of points in designer, moving around a few of the points,
leaving designer and saving.

MAKOMO, 2019-03-25:
I am totally puzzled with this report. The attached alog file is written with v1.5.0 not v1.6.2 as stated. It refers to a time index that does not exists and thus does not load in either version. Instead it fails with the error message (in the error console)

`1 07:08:59.660 Exception: redraw() list index out of range@line 5097`

You should have received this dialog on starting the designer from a profile lacking the CHARGE DROP events

[image]

I just discovered that on clicking OK in this dialog, the designer still starts up, but in a broken environment (just fixed). This might have led to your situation. Could this be possible?

simenmm, 2019-03-25:
Ok, I see. I did get that error message, pressed OK and loaded a new set of
points that included charge and drop. I thought it simply  worked as a
warning that the profile I loaded was missing some data, not that it would
corrupt the profile. I then upgraded to 1.6.2.

MAKOMO, 2019-05-08:
fixed

closed 2019-05-08 by MAKOMO