artisan-roaster-scope/artisan #341: Not sure if this version already comunicates with tiny tonino

elviojunior, 2019-03-27:
## Expected Behavior
after click in scan it fill the form with the color of coffee
## error log 
Version: 1.6.2, revision: ccbc9af4
2019-03-26, 21:01:16
--------------------------------------------------------------------------------
<class 'TypeError'>: 
'<' not supported between instances of 'NoneType' and 'int'
  File "artisanlib\main.py", line 29772, in <lambda>
  File "artisanlib\main.py", line 30859, in scanGroundColor

--------------------------------------------------------------------------------
artisanlib\main.py::<lambda>:29772
	                   _ = False
	                self = <artisanlib.main.editGraphDlg object at 0x00000114E4821D38>
artisanlib\main.py::scanGroundColor:30859
	                   v = None
	                self = <artisanlib.main.editGraphDlg object at 0x00000114E4821D38>

## Actual Behavior
error

## Steps to Reproduce the Problem
just connect the device and click scan

## Specifications

  - Artisan Version: 1.6.2
  - Artisan Build  - Platform Windows
  - Connected devices or roasting machine:

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-05-06:
Thanks for the report and sorry for the late reply. Yes the Tiny is supported and the connection works for me. Did you select the correct serial port in the Port >> Color configuration?

The error message you got is bad and I fixed this by now.

MAKOMO, 2019-05-06:
One more remark: to get the first reading takes a little longer as on some operating systems the initial connect force a reboot of the Tiny. Often it is just faster to type the digits instead of using this connection....

MAKOMO, 2019-05-08:
The communication has be reworked and imporved.

closed 2019-05-08 by MAKOMO