artisan-roaster-scope/artisan #342: Roast Properties, weight -> retain decimals

InvisibleOrdinary, 2019-04-03:
## Expected Behavior

When entering weight that is .0 or .00 that the decimals are retained since that is standard (if you are measuring to tenths, then even if if it's an integer, in order to show that resolution, you show x.0)

## Actual Behavior

Weights drop the trailing 0 below any integer

## Steps to Reproduce the Problem

  1. open artisan and open Roast Properties (cmd-t)
  2. enter weight as xx.00
  3. close & reopen Roast Properties

## Specifications

  - Artisan Version: 1.6.1 (ddd1677)  (+)
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): MacOS 10.14.3
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

DennieSch, 2019-10-06:
I would like to add my support for carrying decimals not only on the Roast Profile page but also throughout the application. For example, carrying decimals into the Batch and Ranking Reports. Web formats of these reports shows only whole numbers for In and Out. Since Excel format reports aren’t currently working yet (I’m the one who brought up the issue with Excel reports) I can’t offer a comment on them. I’m a home roaster with a Behmor so being able to see, print, export, etc. information that includes decimals is very helpful since I work with smaller batch-sizes. 

Using... version = 2.0.0 (a0a1face)

MAKOMO, 2019-10-06:
If you tick "Decimal Places" under menu Config >> Curves, last tab "UI" you will get on decimal on LCDs, in the reports and many other places, like the data table.

The decimals in the Roast Properties dialog, about which this issue is, are rendered without decimals only for whole numbers. So instead of rendering "1.0kg" you see here the more readable "1kg". However, "1.1kg" will render as "1.1kg" always including the decimal. That will most likely not change.

If you need to "export" the recorded temperatures you can as well copy-paste them from the data table (menu Roast >> Properties, last tab "Data") and past them directly into Excel.

Excel reports have been fixed in the current v2.0.1 beta builds https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

DennieSch, 2019-10-07:
Hello  Marko- Thank you for the quick response! (I also saw your response to #399, thanks!)

Actually, I do have "Decimal Places" ticked. I have ticked it and un-ticked it and do see the affects throughout the application. 

To better illustrate what I was trying to describe in my previous comment I've attached a couple jpg files; one for Batch and the other for Ranking. Each jpg has 4 screen captures; one for each of the units of measurement, oz, g, kg and lb.

Batches- 
- kg and lb have decimal values but oz and g do not.

Ranking - 
- kg and lb have decimal values but oz and g do not. 
- Temperature values also do not have decimals (not sure that is as much of a issue as the weights).
- I believe you mention in your reply to #399 that the 'mixed units' in the Load column was resolved but I think the actual values might be suspect for the kg and lb.
- The value for 'AVG Loss' looks a bit suspect too...average Loss of 99% ? :-)

Just trying to share my observations and thoughts to help you make the best application possible. (It is really awesome already... great work!!)

I realize I'm using v2.0 and haven't tried the current pre-release build so some/much of this may be resolved but thought I would share with you just in case. I'll admit I haven't searched the blogs first (which one should do first) but... if you don't mind I'll ask here. Are there any special steps or care to take in installing the current pre-release (uninstall first or install over) and can I roll back to v2.0 easily? If there is a post you want to point me to that would great too.

Cheers- Dennie
[Batches and Ranking Reports - Decimals.zip](https://github.com/artisan-roaster-scope/artisan/files/3695227/Batches.and.Ranking.Reports.-.Decimals.zip)

MAKOMO, 2019-10-07:
Dennie, I think I understood the issues you reported and resolved them correctly for the pre-release  builds.

Pre-release build install like the released versions, just that they are not fully tested. One exception is the macOS platform. As those pre-released builds are not developer signed, they have to be started on their first run via a right-click to the app icon and selecting open to get them accepted by the security framework of macOS.

closed 2019-10-07 by MAKOMO