artisan-roaster-scope/artisan #344: Unable to build artisan.exe from source

fnacer, 2019-04-14:
## Expected Behavior

I'm trying to build artisan.exe from the source code I downloaded from Github. I assumed that I would need to install Python 3.5, add the modules listed in requirements.txt, then run "python setup-win/py py2exe". Build failed until I downgraded jinja to 2.1.8. 

## Actual Behavior

Dozens of missing modules, despite having installed all the ones that could be "pip install"ed. 
[Error.txt](https://github.com/artisan-roaster-scope/artisan/files/3076854/Error.txt)

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: Latest
  - Artisan Build (number in brackets shown in the about box): Latest
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: None
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2019-04-14:
Why not just run from source?  The build process only creates a setup (install) package that when installed runs standalone on another Windows machine.  It offers no advantage if you have Python and the necessary libraries already installed.  

To run from source Install Python 3.5 and use pip to install the libraries in requirements.txt and requirements-win.txt.  Then run "python.exe artisan.py".  Far easier and faster than building a setup package for Windows and then installing before running the program.  

If you really want to tackle building an installer package for Windows you should use pyinstaller rather than py2exe.  Run build-win3-pi.bat.  You will first need to install pyinstaller as well as NSIS.  You also need Microsoft Visual C++ 2015 Redistributable Package (x64), vc_redist.x64.exe located above the source (src) directory.  Hardly worth the trouble if you intend to run the code on your own machine.

fnacer, 2019-04-14:
Would you believe I was so caught up in trying to get a build that I forgot that Python was an interpreted language? 

Running it in Python 3.5 brought up the UI once I installed Phidget22, unidecode, portalocker, persist-queue, and pyserial.

Thank you for your help.

closed 2019-04-14 by fnacer

mehrdad900000, 2024-07-11:
> Would you believe I was so caught up in trying to get a build that I forgot that Python was an interpreted language?
> 
> Running it in Python 3.5 brought up the UI once I installed Phidget22, unidecode, portalocker, persist-queue, and pyserial.
> 
> Thank you for your help.

> ## Expected Behavior
> I'm trying to build artisan.exe from the source code I downloaded from Github. I assumed that I would need to install Python 3.5, add the modules listed in requirements.txt, then run "python setup-win/py py2exe". Build failed until I downgraded jinja to 2.1.8.
> 
> ## Actual Behavior
> Dozens of missing modules, despite having installed all the ones that could be "pip install"ed. [Error.txt](https://github.com/artisan-roaster-scope/artisan/files/3076854/Error.txt)
> 
> ## Steps to Reproduce the Problem
> 
> ## Specifications
> * Artisan Version: Latest
> * Artisan Build (number in brackets shown in the about box): Latest
> * Platform (Mac/Windows/Linux + OS version): Windows 10
> * Connected devices or roasting machine: None
> 
> Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file. Please attach any relevant Artisan *.alog profiles.
> 
> _Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

Hello good time
Sorry
You could get exe output from the source file