artisan-roaster-scope/artisan #349: Support changing the Qt5 style

tanius, 2019-04-30:
## Expected Behavior

As a Qt5 application, `artisan` is kind of "expected" to let users choose / configure the user interface style in [the ways this is possible in the Qt5 framework](https://wiki.archlinux.org/index.php/qt#Qt5):

1. With a command-line argument, for example:

    ```
    artisan -style=cleanlooks
    ```

2. With an environment variable, for example:

    ```
    QT_STYLE_OVERRIDE=cleanlooks artisan
    ```

3. With the application [`qt5ct`](https://sourceforge.net/projects/qt5ct/), for example as follows:

    ```
    QT_QPA_PLATFORMTHEME=qt5ct qt5ct
    ```

    Now make some UI style changes. Then:

    ```
    QT_QPA_PLATFORMTHEME=qt5ct artisan
    ```

I tried all of these alternatives also with a generic Qt5 application ([`tom-ui`](https://github.com/jansorg/tom-ui)), where the UI style changes were effective.

## Actual Behavior

None of these change the user interface look & feel of Artisan in any way :(

It seems that earlier versions of Artisan had a configuration setting to select the UI style (as referenced in #159 ), however, I could no longer find this in the user interface. The configuration file `~/.config/YourQuest/Artisan.conf` contains a key `[General] appearance=`, but when setting this to any Qt5 style name (such as `cleanlooks`, `motif`) and starting Artisan, nothing changes. Additionally, Artisan saves this configuration key with an empty value when exiting.

## Specifications

  - Artisan Version: 1.6.2
  - Artisan Build: 9d1202a
  - Platform: Lubuntu Linux 18.10 (Ubuntu with LXQt desktop environment)

tanius, 2019-04-30:
**P.S.:** The reason why I am trying to change the Qt5 UI style is because Artisan shows up like this on my system, with a very small fixed-width menu font. I did not see this issue in any screenshots of Artisan on the web, so I thought it may be due to some peculiarities of my system configuration (though it does not apply to other Qt applications I have). So I am trying to find a way to adapt that font style and size.

[image]

MAKOMO, 2019-05-06:
The settings moved under Config >> Curves, last tab "UI". It might not help with your font issue. Artisan is not doing anything specific regarding the Qt app font.

closed 2019-05-06 by MAKOMO

tanius, 2019-05-06:
Thank you for the pointer, I will have a look there.

Independent of my system's font problem, you might want to keep this issue open as a feature proposal to support the `-style` and `QT_STYLE_OVERRIDE` mechanisms of changing the Qt5 style. It's pretty standard behavior in all the Qt applications I know.