artisan-roaster-scope/artisan #350: artisan window 32bit

merapicoffee, 2019-05-03:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. i need artisan for windows 10 32bit sir
  2. you can contact me at coffeemerapi@gmail.com
  3. thnak you very much for your attention

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-05-03:
There are no 32bit builds anymore. Artisan v1.1.0 was the last version that featured a 32bit Windows build.

https://github.com/artisan-roaster-scope/artisan/releases/tag/v1.1.0

Of course you any version from source using a 32bit Python and Qt installation.

closed 2019-05-03 by MAKOMO

merapicoffee, 2019-05-03:
Thank you very much 🙏🏾