artisan-roaster-scope/artisan #352: Random data drops due to USB loss

DmitryBurstein, 2019-05-13:
Sorry for filing an issue with an old version, but it was the last one available for 32-bit architecture of my roasting netbook.
I experience constant and seemingly random data losses occurring due to dropped USB connection to the thermometer. The connection is restored all by itself after seemingly random period of time, but then the process repeats.
I noticed the connection is dropped only during the actual logging phase - after pressing the "Start" button, but never during the "On" phase prior to it. Last time (see dmesg log attached) I left the program "On" (but not started) for more than an hour - and not even once during this time were the connection lost. However after the roast had started, the connection was lost and restored multiple times (see the alog and dmesg attached).
Any help in mitigating this problem would be highly appreciated.

## Specifications

  - Artisan Version: 1.1.0
  - Artisan Build (number in brackets shown in the about box): 2
  - Platform (Mac/Windows/Linux + OS version): Linux Fedora 19 (kernel 3.14.27-100.i686)
  - Connected devices or roasting machine: Mastech MS6514

## Files attached

[artisan-settings-netbook.txt](https://github.com/artisan-roaster-scope/artisan/files/3173130/artisan-settings-netbook.txt)
[batch_90_19-05-13_1244.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/3173131/batch_90_19-05-13_1244.alog.txt)
[dmesg.txt](https://github.com/artisan-roaster-scope/artisan/files/3173132/dmesg.txt)

MAKOMO, 2019-05-13:
Sorry, we have no resources to work on older versions. You are here on you own. I remember that the Mastech driver within Artisan was reworked at one point. Not sure if this was before or after v1.1 got released. The Mastech uses a rather weak communication protocol that tend to produce comm errors. Note further that your issues might also be hardware related as ground loops can trigger data losses too and those occur more often if beans bump against TCs after CHARGE.

DmitryBurstein, 2020-05-26:
This issue should of course remain closed, but I have found a solution and wanted it to be documented somehow - in case anyone else will encounter it in the future. Moreover, I think the problem is not specific to the older versions of Artisan.
You were absolutely right to point out it was probably a hardware problem with Mastech device/protocol, and your mentioning ground loops gave me an idea to connect the USB cable via ADuM3160 based USB isolator - it's a small and cheap ($5-7) widget I got online. Since I started using it - almost 100 roasts with no data loses whatsoever!
So thanks for the idea, and may be you should mention this possible problem/solution on your devices->meters page.

MAKOMO, 2020-05-26:
Thanks Dmitry, I will add such a note. However, note that there is no difference in the sampling between ON and START modes on the Artisan side. Thus I believe that the issue you observe must be related to the beans bumping on the probe.

closed 2020-06-02 by MAKOMO