artisan-roaster-scope/artisan #354: Enhancement Requests: Run a script when an alarm occurs?

DavidFW1960, 2019-05-19:
So this isn't a bug but an enhancement request.

I setup my Roaster in my shed and often, it overuns the temperature at which I would normally charge it.

I am running a RaspberryPi headless with VNC and I control it from a PC in my house which works really GREAT. But the alarm does not work over the VNC connection and I might see the popup only if the VNC screen is in the foreground but in reality I often miss it and have to put the roaster back to cool and try again...

It would be really great if an alarm could trigger an external script (or even better would be to publish to a MQTT topic). I could then use other running software (Home Assistant) to send me a notification when tat event is triggered. This would be a great enhancement.

## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-05-19:
The alarm action below "Pop Up" is "Call Program".  This features is for example used to create "Talking Alarms" on Windows <https://artisan-roasterscope.blogspot.com/2017/12/talking-alarms.html>. Alarms can also trigger custom buttons that also can run scripts. Further, you might want to run the WebLCDs which is a WebService based remote Web display of the TIME, ET, and BT LCDs that also has an option to show alarm popups in the connected Web Browser (eg. on your iPhone).

closed 2019-05-19 by MAKOMO

DavidFW1960, 2019-05-30:
Thanks Marko.. I missed that - talking-alarms wasn't quite what I was expecting to look for lol!
Anyway, I wrote a shell script to publish to MQTT on certain events and configured those scripts as an external program and it works great. Thanks for the pointer in the right direction.