artisan-roaster-scope/artisan #355: Slow to open the Software

reizao, 2019-05-23:
## Expected Behavior

Until yesterday the software opens in about 5 seconds

## Actual Behavior

The software takes about 2 minutes to open

## Steps to Reproduce the Problem

  1. click the icon
  2. wait
  3. 

## Specifications

  - Artisan Version:1.6.2
  - Artisan Build (number in brackets shown in the about box): 9d1202a4
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Phidgets 1048
[king.zip](https://github.com/artisan-roaster-scope/artisan/files/3211787/king.zip)
[king.zip](https://github.com/artisan-roaster-scope/artisan/files/3211792/king.zip)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2019-05-30:
Is this still happening?  Did anything change on your Windows system or your Artisan setup right before this started happening?  One thing to try - since you have your custom settings already saved in king.aset, try Help>Factory Reset and restart Artisan.  Any difference in the startup time?  You can restore your custom settings with Help>Load Settings.

reizao, 2019-05-31:
I will try

reizao, 2019-06-02:
When I changed to factory defaults it starts quick, but when I read the custom settings it slows again. Is there any way to disable artisan plus?

MAKOMO, 2019-06-02:
To disable artisan.plus just click on the blue plus icon in the upper left corner and the client gets logged out. Please report here if this helped. In that case I would love to debug and fix what is wrong on the plus mode. If this did not help, we have to find the issue in other parts of Artisan.

reizao, 2019-06-03:
[image]
There is no plus icon on my screen.

MAKOMO, 2019-06-03:
Then you are using a version without the plus module and the slow down cannot be related to plus. You could check if you have the WebLCDs enabled. Those start a whole WebServer within Artisan and this might take ages on slow machines with low memory. The WebLCDs can be enabled/disabled under menu Config >> Curves, last tab "UI".

reizao, 2019-07-02:
Well, WebLCDs are not enabled. The machine have 16 GB Ram installed, 512 SSD Drive, Intel I7 from Dell. But the internet where is installed is very slow.

reizao, 2019-07-03:
I've installed 2.0.0 version and returned to factory defauts, but it still slow to open

MAKOMO, 2019-07-05:
Hm. Before you stated that after a factory reset Artisan would start fast again. As this is not reported by anyone else, this might be just an issue on your local machine. Did you try to run Artisan on another machine?

MAKOMO, 2020-04-04:
stall

closed 2020-04-04 by MAKOMO