artisan-roaster-scope/artisan #356: Using the Event Sliders to Control Linear Actuators

SACoffee, 2019-05-30:
I don't know how useful this would be for other people, but Canadian company Actuonix makes some super nifty linear actuators with feedback controls that allows users to adjust the linear actuator with a simple configuration software that they provide.  What this means is that roasters who have to manually adjust airflows or gas servos can attach these things to the linear actuator and use the program to precisely adjust those values. 

**The problem:**
We roast on two Sasa Samiac roasters with Weishaupt burner control units.  These BCUs are usually single or dual stage (one or two settings) with no way to control the flames (other than starving the gas check system with lever ball valve or using an electric screw driver to adjust a set screw).  Adjusting these is hard because you a) have to go to the back of the roaster to do so and return to the front to see the effect on artisan b) starving the gas too much causes a reset to the burner control.  

Also, the dampener is in the back of the roaster so any adjustment made to the damper means you cant see the result on the artisan screen and its just a hassle to walk back and forth especially during more crucial phases.

**The solution:**
I disconnected the weishaupt servo from the burner actuator mechanism body but kept it installed on the machine so that when the system runs the servo check all is good to go.  To the gas control mechanism I installed the linear actuator, which is controllable by the computer. Instead of a static setting I used to have from the single stage servo I am using the linear actuator to actually control the gas flow.  It works freaking awesome!  Only problem is I have to run the LAC configuration program alongside the artisan program and if I'm controlling both the airflow and the gas it covers much of the artisan screen.

**The feature I would like to see:**
The LAC configuration software integrated into the artisan software by use of the sliders.  This way I would not need to make the LAC window always in front, blocking the screen (worse with 2 open at once).  Also, if the sliders controlled the actuators, any changes in the airflow or gas levels would be logged!!

I see this as a super useful modification to older roasters that never had proper controls installed on them to change gas and airflow on the fly.

MAKOMO, 2019-05-30:
Dear SACoffee,

Artisan v1.6.2 added support for Phidgets RC controllers that can drive the Actuonix RC linear actors like this one

https://www.phidgets.com/?tier=3&catid=25&pcid=22&prodid=406

See the section under "RC Servos" in the following post, on how those can be controlled directly via Artisan sliders

https://artisan-roasterscope.blogspot.com/2019/03/artisan-v161.html

Hope this helps

closed 2019-05-30 by MAKOMO

SACoffee, 2019-06-04:
Dear MAKOMO,

Thanks!  Any chance for support for the Actuonix LAC board to be added?  I was thinking maybe Artisan can control the LAC Configuration Utility, but I don't have any programming knowledge.  Also, from the blog it looks like currently only one servo can be used at a time (so no way to control both the gas and airflow from artisan).  Is that correct?'

Thanks again!
Sami

MAKOMO, 2019-06-04:
If the community is not providing support for the Actuonix board it is unlikely to get every direct support. Note that you can always drive it via your own script triggered by Artisan without any need to change anything in Artisan. Driving another app is often not a good solution. Does the Actuonix board support at all open APIs that are available cross platform?

Buying one of the supported Phidget boards is some order of magnitude cheaper than to implement support for the Actuonix board. As documented, those Phidgets boards all RC ports are supported which is up to 16 on the larger one.

SACoffee, 2019-06-04:
Appreciate the help on this!