artisan-roaster-scope/artisan #357: Register for Plus not working

ibrubeer, 2019-06-04:
## Expected Behavior
- Get verify email, click on link to verify

## Actual Behavior
- Verify link takes me to a page that says "email verify failed: : wrong email verification link"

## Steps to Reproduce the Problem

  1. Go to the Plus web page (https://artisan.plus/en/)
  2. Register 
  3. Click on link in email
[image]

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-06-04:
Thanks for the notice. This got resolved. We will start to enable registered accounts after we returned from the World-of-Coffee event in Berlin.

closed 2019-06-04 by MAKOMO

ibrubeer, 2019-06-04:
Will I get a new email with a verify link that works? I used my preferred
email address already, and the system knows it now. When do you return from
Berlin?

Thank you!
Dennis

MAKOMO, 2019-06-04:
I checked, your email is verified. We return on Sunday from Berlin and will start activating accounts on Monday.

ibrubeer, 2019-06-04:
Ok, great! Thank you. Enjoy the rest of your trip in Berlin!

Dennis