artisan-roaster-scope/artisan #358: Suporte a automção e controle

wbh16, 2019-06-05:
We have performed some tests and use of the artisan for control and automation of the equipment that make up the roaster. When we use modbus there is a delay because we use the same medium for communication. I would suggest using another channel such as the MQTT protocol used in IOT (internet of things). Or another technique that optimizes the use of automation controls by artisan.

wbh16, 2019-06-05:
Temos efetuado alguns testes e uso do artisan para controle e automação dos equipamentos que compoem o torrador. Quando utilizamos modbus existe um delay, porque usamos o mesmo meio para comunicação. Gostaria de sugerir o uso de outro canal como por exemplo o protocolo MQTT usado em IOT (internet das coisas). Ou outra tecnica que otimize o uso dos controles da automação pelo artisan.

Filipe-Medeiros, 2019-06-05:
I'm from Brazil, I'm together with @ wbh16 in the development of the communication hardware via ModBus mentioned, where we control the gas roaster inside the Artisan softwere itself, but this dalay that commented wbh16 exists in the commands, affects the usuality of controls in roasters, please if you can give special attention to this subject, it will help us a lot to grow along with Artisan.
Along with this suggestion I would like to thank and greatly praise the other functions of the software as well as helping in the development of communication hardware I am a producer of artisanal coffee here in Brazil and I use Artisan in my toasts. Very good.

MAKOMO, 2019-07-05:
MODBUS communication via TCP or UDP is quite instance. I observed the communication between Artisan and a Probat Probatone and there is a massive amount of commands exchanged per second without any observable delay.

Filipe-Medeiros, 2019-07-15:
Hello Friend! Is there any way to eliminate the existing delay
in the MODBUS communication? I did not quite understand your comment.
Please give us a solution or a way to solve this problem. THANK YOU.

Filipe-Medeiros, 2019-07-15:
Hello Friend! Is there any way to eliminate the existing delay
in the MODBUS communication? I did not quite understand your comment.
Please give us a solution or a way to solve this problem. THANK YOU.

> A comunicação MODBUS via TCP ou UDP é bastante instável. Observei a comunicação entre o Artisan e um Probatone Probat e há uma enorme quantidade de comandos trocados por segundo sem qualquer atraso observável.

Filipe-Medeiros, 2019-07-15:
> Olá amigo! Existe alguma maneira de eliminar o atraso existente 
> na comunicação MODBUS? Eu não entendi muito bem o seu comentário. 
> Por favor, nos dê uma solução ou uma maneira de resolver este problema. OBRIGADO.
> 
> > Uma comunicação MODBUS via TCP ou UDP é bastante instável. Observar uma comunicação entre o Artisan e um Probatone Probat e a enorme quantidade de comandos trocados por segundo, sem qualquer atraso observável.

Filipe-Medeiros, 2019-07-15:
Hello! there would be the possibility of deploying an MQTT communication protocol as suggested by the user @ wbh16, this type of protocol is very stable and fast communication responses. Give us information on the subject, and if we can help in any way we are available. thank you.

MAKOMO, 2019-07-15:
There are no plans to add MQTT to Artisan. If you need MQTT you have to add it yourself. The source code is available. If you think your work is worse to be shared to others turn it into a PullRequest and we will review and potentially integrate your changes.

closed 2019-07-15 by MAKOMO

Filipe-Medeiros, 2019-07-15:
Cool! If I can develop something in MQTT I will share with the artisan team. About your test by watching Probat, what were the results found? What is the solution for MODBUS communication delays? What is Artisan's fastest and most stable method of communication?
thank you.

MAKOMO, 2019-07-15:
Stablity and speed of communication is only a problem with slow (9600 baud) serial MODBUS or serial2USB meters.  But even there problems arise only if the Artisan sampling interval is <3sec.