artisan-roaster-scope/artisan #360: Roast/Cup Profile: value keep changing to max or min if double clicked

bigl00z3, 2019-06-09:
## Expected Behavior
in Roast/Cup Profile.. When double clicking on a value for the spider chart, the value should move by 2 steps

## Actual Behavior
When double clicking, the values keeps changing until reaching either the lowest (0) or the highest (10) value (takes around 14 seconds to move from 0 to 10).

## Steps to Reproduce the Problem

  1. Open Roast file 
  2. Open menu Roast/Cup Profile.. 
  3. Double click on the arrow to increase a value.

## Specifications

  - Artisan Version: tested in Artisan 1.6.99 and Artisan 2.0.0 
  - Artisan Build (number in brackets shown in the about box): tested in (d8854fe) and (a0a1fac)
  - Platform (Mac/Windows/Linux + OS version): OS X
  - Connected devices or roasting machine: Currently none. I'm just editing a roast after cupping it.
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

[Artisan files.zip](https://github.com/artisan-roaster-scope/artisan/files/3269383/Artisan.files.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-07-05:
Hm. Dave and myself cannot duplicate this on our RPis (nor on Windows and Mac OS X). This might be caused either by the outdated Qt 5.7 GUI lib bundled with Artisan for Raspian Stretch or we just do not understand clearly the issue.

bigl00z3, 2019-07-05:
It’s actually on the Mac that I have the issue (where I edit the cup profile, generally the next day when I’m tasting it) not on the Raspberry

I did a quick video of it:

https://bigloose.com/n/index.php/s/KswfACsKppK4k75

Nicolas

MAKOMO, 2019-07-05:
Wow! Now I can duplicate the issue at least on one of my machines. It does not happen always, but it happens. I have also an idea why. Very strange chain of events... I'll try to fix this with a work around.

MAKOMO, 2019-07-05:
Commit 53c1500ec1a13ed1c990e7b123486dccbc90937f somehow improves on the situation caused by slow redraw of the canvas but does not fully resolve it. A tripple click still leads to this behaviour.

Related:
https://stackoverflow.com/questions/41568990/how-do-i-prevent-double-valuechanged-events-when-i-press-the-arrows-in-a-qspinbo

closed 2019-07-05 by MAKOMO