artisan-roaster-scope/artisan #361: Allow user to set event button sizes (Enhancement)

Brewzologist, 2019-06-09:
## Expected Behavior
!!! First off:  Thank you for a great roasting application !!!

This is an enhancement request:  Allow the user to choose sizes for event buttons.  (I cannot find any way to do this in the documentation).

## Actual Behavior
On my laptop, the event buttons are small which makes them difficult to use during a roasting session.  See attached screen shot.  I would prefer the ability to make them larger (like the default buttons).

[Artisan Event Buttons.zip](https://github.com/artisan-roaster-scope/artisan/files/3269644/Artisan.Event.Buttons.zip)

## Steps to Reproduce the Problem

  1.   See attached
  2. 
  3. 

## Specifications

  - Artisan Version: v2.0  (Also the same in v1.x releases)
  - Artisan Build (number in brackets shown in the about box): N/A
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine:  Huky using Phidget 1048, and Arduino for fan/gas control
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

arkenamun, 2019-07-09:
A short note to also add another usage example as a Windows 10 user on the change to user experience in button arrangement from v1.x, in case this helps with this report.

I appreciate an enhancement has been made in respect of this to allow event button sizes to be set. That could well be very useful and a comprehensive resolution. Thank you.

I have 21 buttons visible  in row 1. Under v1.x ,and with 30 buttons max per row, they fitted the full width of the bottom section of the scope window (larger/wider/full width). I would speculate that maybe there was some scaling that has been lost as a result of the improved grouping/spacing functionality, or maybe the buttons are just smaller. In either case, size options would likely be a great solution.

Thanks, Brad
PS
1) Note I am also increasing some button sizes by padding with spaces
2) My full .aset is available under issue 373

[image]

MAKOMO, 2019-09-09:
Added the possibility to choose the extra event button sizes to be either "small" (as in v2.0, but with a slightly larger font), "large" (size of the default event buttons), or "tiny" (smaller than in v2.0 to allow a large number of buttons in one row). Commit 048d908aef1580e258994bc73528dbbc516c5a34.

closed 2019-09-09 by MAKOMO