artisan-roaster-scope/artisan #362: PID Control:Record Air events

ZhendongRen, 2019-06-12:
## Expected Behavior
If the action of the air slider can record the event (PID negative control air), it would be great！

## Actual Behavior
Currently, by manually adjusting the Air sliding interface, send an RC command to change the Air.
Air slider operation is interpreted as temperature and recorded in the event.

Try to negatively control the air through the PID, the PID will automatically adjust the Air slider, so that BT repeats the background, but these actions will not be recorded in the event.

## Steps to Reproduce the Problem

  1. Config>>Events>>menu Sliders
  2. Add the Air slider and the slider value is interpreted as temperature
  3. Negative control of air through PID

## Specifications

  - Artisan Version:2.0
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):Windows 10
  - Connected devices or roasting machine:Bideli 1KG electric coffee roaster
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-07-05:
I do not understand this setup nor the issue. Sorry. Attaching your Artisan settings as requested would help.

What do you mean by "Add the Air slider and the slider value is interpreted as temperature"? The Artisan sliders are event sliders and they generate events of the selected value, not "temperatures" what ever that means!?

The PID is taking as input the selected temperature channel and can operate event sliders as target. I just checked, but slider events are recorded also if the slider was automatically moved by the PID (also under negative control).

Please clarify!

ZhendongRen, 2019-07-11:
The PID control slider moved to the 0 position, but the event was not recorded.
Nearby are profiles and screenshots.

[PID.zip](https://github.com/artisan-roaster-scope/artisan/files/3380863/PID.zip)

ZhendongRen, 2019-08-06:
[PID.zip](https://github.com/artisan-roaster-scope/artisan/files/3472371/PID.zip)

Attachments are profiles and screencasts，Manually adjusting the slider will log the event, but the PID control slider will not log the event.

ZhendongRen, 2019-08-14:
> 我不明白这个设置也没有问题。抱歉。根据要求附加您的Artisan设置会有所帮助。
> 
> “添加空气滑块和滑块值被解释为温度”是什么意思？Artisan滑块是事件滑块，它们产生所选值的事件，而不是“温度”，这意味着什么！？
> 
> PID将所选温度通道作为输入，并可将事件滑块作为目标。我刚检查过，但如果滑块由PID自动移动（也在负控制下），也会记录滑块事件。
> 
> 请澄清！

https://youtu.be/Rtt0gkAUc5k
This is the recording screen. The attachment in the previous reply is the configuration file.

MAKOMO, 2020-04-04:
Yes, changes generated by the PID are not registered as events as usually there are too many. This is a design decision and might not change in the near future. Sorry.

closed 2020-04-04 by MAKOMO