artisan-roaster-scope/artisan #364: retain view settings when closing

InvisibleOrdinary, 2019-06-16:
## Expected Behavior

View/Show Readings should retain setting on closing/etc… 

## Actual Behavior

The LED / extra devices data on the left don't stay up when closing artisan

## Steps to Reproduce the Problem

  1. select Menu/View/Readings
  2. close artisan
  3. re-open artisan

## Specifications

  - Artisan Version: Artisan 2.0.0 (a0a1fac) 
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14.5
  - Connected devices or roasting machine: various

[190616_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/3294406/190616_artisan-settings.aset.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

InvisibleOrdinary, 2019-06-16:
Actually, this seems to be reset anytime the view changes, for instance if you select menu/Roast/Switch Profiles (cmd-W) the are turned off as well

MAKOMO, 2019-07-05:
Once you press ON or START they show up and show the values of your input devices. Once you press OFF they hide away automatically.

Why you want to stare at a row of empty/zero LCDs on opening Artisan or stopping the data gathering?

InvisibleOrdinary, 2019-07-11:
(slight correction: I meant the readings on the right, but I think that was understood)

I would like to see them for several reasons: 

- So that the screen doesn't jump around when I start the roast - its visually distracting, and unnecessary. 
- So that I can see them when working on adding / modifying extra devices. It's pretty cumbersome to have to add the formulas, then start a roast just to make sure the order is what I want.
- So that when I check an option to show a reading (even if it is null), that the reading is shown... when I check that option.
- I'm not sure how much this affects the overall speed of artisan, but it feels that the less it's doing (like updating a value vs. redrawing the screen to add the LEDs) the quicker it will be when pressing start. At least it seems this would be true. 

… is there a reason why it's turned on and off with the recording that aids in the functioning of Artisan?

MAKOMO, 2019-08-27:
Well, Artisan tries to hide elements that don't cary productive information to maximize the screen estate available for the roast graph at any time.

Still, I follow your suggestion and changed this behavior (86cebe76ab7d38c3d6fb489843dc7b2125c16a5c).

Thanks for the suggestion,
Marko

closed 2019-08-27 by MAKOMO