artisan-roaster-scope/artisan #366: Need Helping Running Program Connected to Phidgets

FloraRoasting, 2019-06-24:
## Expected Behavior
I am trying to run a program that reads the temperature from my Phidgets1045 IR Temperature Sensor. My goal with the program is to be able to organize the samples and average only the lowest 10 samples of the approximately 30 samples per second. I have scoured the internet and do not know code well enough to write it myself. Any support or a start in the right direction is greatly appreciated. 

## Actual Behavior

I get an error with my program not giving artisan the temperature. 

## Steps to Reproduce the Problem

  1. On 
  2. Start
  3. 

## Specifications

  - Artisan Version: 2.0.0
  - Artisan Build (a0a1fac):
  - Platform: Mac
  - Connected devices or roasting machine: Phidgets 1045_1B IR Temperature Sensor
  
[JacobsRoastingSettings.alog.zip](https://github.com/artisan-roaster-scope/artisan/files/3321944/JacobsRoastingSettings.alog.zip)

This is what I have so far for my program build.

import sys
import time
import traceback

from Phidget22.Devices.TemperatureSensor import *
from Phidget22.PhidgetException import *
from Phidget22.Phidget import *
from Phidget22.Net import *

try:
    from PhidgetHelperFunctions import *
except ImportError:
    sys.stderr.write("\nCould not find PhidgetHelperFunctions. Either add PhdiegtHelperFunctions.py to your project folder "
                      "or remove the import from your project.")
    sys.stderr.write("\nPress ENTER to end program.")
    readin = sys.stdin.readline()
    sys.exit()
    

ch0 = TemperatureSensor()
ch0.setDeviceSerialNumber(424909)
ch0.setChannel(0)
ch0.openWaitForAttachment(5000)
ch0.setTemperatureChangeTrigger(0)

ch1 = TemperatureSensor()
ch1.setDeviceSerialNumber(424909)
ch1.setChannel(1)
ch1.openWaitForAttachment(5000)

ch1.setTemperatureChangeTrigger(0)

ch0.getTemperature()
ch1.getTemperature()

Print 
void onTemperatureChange(
TemperatureSensor self,
float temperature
)

ch0.close()
ch1.close()

MAKOMO, 2019-06-24:
Please keep this channel free for reports on issues in the Artisan roast logging software!

closed 2019-06-24 by MAKOMO

FloraRoasting, 2019-06-24:
My apologies Marko as I could not find any contact info to reach out for
help. Do you know where I can ask for assistance with this?

Thank you for your time and again I apologize about posting in the wrong
area.

Jacob