artisan-roaster-scope/artisan #367: Artisan+ Save setting icon

InvisibleOrdinary, 2019-06-26:
## Expected Behavior
…not sure if this is where we should be putting the online issues or not… please redirect if this is the wrong place -)

The Icon for saving your settings when changing is easily recognizable

## Actual Behavior

The Icon for saving settings looks almost identical to the typical icon for downloading something

MAKOMO, 2019-07-05:
I was confused and didn't understand exactly what you are talking about. Did you talking about the Artisan client or the artisan.plus service / Web app?

Now I got it. In the artisan.plus Web Apps account settings dialog (absolutely ok to report issues here, we do not have a separate tracker for plus) there is now the possibility to change your password and a corresponding icon to save the changes and there is also the same icon used under "Download your Data". Confusing indeed.

We are struggling to find a good "save" (material) icon. The alternative to what we have now is a diskette symbol which might not ring a bell for some of the younger users.

However, we could replace the "Download your data" with a the material cloud download icon from here https://material.io/tools/icons/?search=cloud&style=baseline

InvisibleOrdinary, 2019-07-05:
I think this is your call… the cloud d/l seems wrong for saving, but I wouldn't be confused by the disk (I'm old enough to have had 5.25" floppys at one point, so…)
Are you locked into using only the material supplied icons? could you modify / add?

MAKOMO, 2019-07-05:
Yes, the cloud download would be used for the download of the data set which is a cloud dowload. That would be fine.

Yes we can modify and add, just the style should be similar to fit the other icons. Still I haven't seen a convincing shape yet. The disk looks wrong to me too as you never store on a floppy disk (I even saw the 8" floppies;) those days, besides that this action is mostly just a "store" action to an internal state. 

Hm. For now we will use the cloud download icon for the cloud download, keep what we have for save. Of course we could use text buttons, but that would "pollute" the translation process unnecessarily. 

I keep this open for the moment.

InvisibleOrdinary, 2019-07-06:
Ok. When I have more time in the next several weeks, I can send a few variations for you to choose from. Like that you prefer to keep it visual... and will keep it close to the material direction.

Cheers

MAKOMO, 2019-07-06:
Cool! Thanks a lot!

smaca69, 2019-07-08:
what about the cloud_done icon to use as the save?

MAKOMO, 2019-07-17:
Yes, now we established the cloud_done icon to save/export from the cloud and the tick icon for all OK buttons. That should resolve this confusion.

closed 2019-07-17 by MAKOMO