artisan-roaster-scope/artisan #369: Artisan+ / Can't save new green in store

InvisibleOrdinary, 2019-06-26:
## Expected Behavior

Can save entered beans in store.

## Actual Behavior

the screen does nothing, and if I navigate away, all entries for that bean are lost, including adding a new producer… I had entered on soe time before, and had no issue with updating it's info through interfaces (sections of the site) but this won't seem to save at all

MAKOMO, 2019-07-05:
Hm. Does this issue persist. Could you show a screen shot of this or anything that helps us to duplicate this? Thanks!

InvisibleOrdinary, 2019-07-05:
Hmm, I tried to reproduce, but can't so maybe this was a glitch from the account issue?

MAKOMO, 2019-07-05:
We might have fixed this in one of the recent deploys. I tried too, but couldn't duplicate this anymore. I close this for now

closed 2019-07-05 by MAKOMO