artisan-roaster-scope/artisan #370: Displaying ROR value at top of screen in relation to cursor position

FCRTurku, 2019-06-27:
## Expected Behavior
At the top of the screen x(time) and y(temp) are displaying value according to where the curcor is located on the graph.  It would be helpfull to have the ROR axix display here too.

## Actual Behavior
Only x(time) and y(temp) are displayed.

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:2.0.0
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Windows
  - Connected devices or roasting machine: Loring S7
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roundfile, 2019-06-27:
There is an existing the keyboard shortcut 'd' that will toggle the displayed y value between Temp and Delta.  You will need to move the pointer after the touching 'd' to update the display.

roasterdave, 2019-06-27:
There is an existing keyboard shortcut 'd' that toggles the displayed y value between Temp and Delta.  You may need to move the pointer slightly to update the display after hitting 'd'.

closed 2019-06-27 by roasterdave