artisan-roaster-scope/artisan #371: Roast Properties "Title" Font Color problem.

PapasCup, 2019-07-01:
## In the Roast-Properties-Title, the "Title" data entry font should not follow what is set into the Config-Colors-Graph (tab)-Title color.  It is currently doing that and if the Config-Colors-Graph (tab)-Title color is set to "white", the Roast Properties Title box is unreadable (white font against a white background in the data entry field box).  White as a Graph Title should be a valid choice, but in the Roast-Properties-Title data entry box, the Font should always be Black and not be impacted by the User's color choice within the Config-Colors-Graph (tab)-Title color choice.

## If you have the Config-Colors-Graph (tab)-Title set to #ffffff (white), the Font color over in the Roast Properties "Title" box (for title entry) will be also white against a white background for that text box.  That makes it unreadable in the Roast Properties Title data entry box.

## Steps to Reproduce the Problem

  1. Go to Config-Colors-Graph (tab)-Title and set the color to #ffffff (white)
  2. Then go to Roast-Properties-Title (field) and the title will be blank (white font against white background).
  3. 

## Specifications

  - Artisan Version:  2.0.0 
  - Artisan Build (number in brackets shown in the about box):  (a0a1face)
  - Platform (Mac/Windows/Linux + OS version): Windows 7 and Windows 10
  - Connected devices or roasting machine:  Aillio Bullet R1 (ver 1.5)
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2019-07-01:
Thanks for reporting.  Confirmed on Windows.  Investigating the other platforms.

MAKOMO, 2019-07-02:
Fixed by 33f9979e19aac65b015027e51fe4b471aa98059c

closed 2019-07-02 by MAKOMO

PapasCup, 2019-07-02:
Hi Marko, thank-you for fixing the problem.  I’m new to GitHub and this process.  Do I wait for a future release of Artisan 2.0.x or is this fix available in some sort of interim release?  If there is a web-page that I missed that answers these questions or where interim releases are help, I’d appreciate knowing it.

 

Rick

 

Fixed by  <https://github.com/artisan-roaster-scope/artisan/commit/33f9979e19aac65b015027e51fe4b471aa98059c> 33f9979