artisan-roaster-scope/artisan #372: Overwrite batch counter from settings file; No.... Does not overwrite batch prefix.

arkenamun, 2019-07-03:
## Expected Behavior

Batch prefix may continue to be loaded from settings file, even if we choose not to overwrite batch counter from the settings file.

[image]

## Actual Behavior

If we say 'NO' to 'Overwrite your current batch counter h2839 by c2749 from the settings file?' then neither the batch (2839) nor prefix (h) are overwritten.

That seems reasonable and logical according to the question, yet may not be practical if the user uses the prefix to record which settings file/type of roast is being produced in this session. The user may be using multiple settings files, yet still wishes to keep each batch number sequential and unique irrespective of type of coffee roasted.

I think it makes sense that the prefix is always loaded as this is a static setting of the settings file that, along with the other static settings, is chosen by the user. Unlike the batch number which is expected to change dynamically with roasting.

N.B. For background, in my use case, I have 2 settings files. The main difference is in the Axes settings, one has higher time and temperature for a hotter, longer, dark roast (2nd crack). The other setting file is for 3rd wave lighter roasts. But also some of my settings for over-temperature alarms and aspects of roast automation are different. 

The, recently added, display of batch number during roasting has the prefix and is great reassurance at the start of roasting that we have remembered to open the correct settings.

## Specifications

  - Artisan Version: 2.0.0
  - Artisan Build (number in brackets shown in the about box): a0a1face
  - Platform (Mac/Windows/Linux + OS version): W10
  - Connected devices or roasting machine: Phidgets and Modbus

MAKOMO, 2019-07-05:
Thanks for your report. While I could think of users that might complain if the prefix does change with the loaded settings, I will follow your proposal and treat the batch prefix as any other application setting. Just the pure batch counter (number) will be treated special then.

MAKOMO, 2019-07-06:
Fixed by Commit 003724c69a5fca324deb0ec64df0ce0b43342290

closed 2019-07-06 by MAKOMO