artisan-roaster-scope/artisan #376: Artisan Command (Button) tare(3) not functioning

arkenamun, 2019-07-09:
## Expected Behavior

Trying to write current LCD reading to T3 on channel3/first extra device with command tare(3)

## Actual Behavior

T3 apparently unchanged

## Steps to Reproduce the Problem

  1. Create button Artisan Command tare(3)
  2. Press button (39 in my jpeg below)
  3. Compare with right click on LCD
4. Double-check configuration vs v2.0 blog post, seems correct.

## Specifications

  - Artisan Version: v2.0.09
  - Artisan Build (number in brackets shown in the about box):a0a1face
  - Platform (Mac/Windows/Linux + OS version):W10
  - Connected devices or roasting machine:Phidgets 1018 888 for this channel
  
[image]

[image]

MAKOMO, 2019-07-11:
Sorry. On introducing the Artisan Command "tare" I also introduced 2 typos that prevented it to work properly. This should be fixed by Commit 9d16347dadec2397b5706e95aeeeefb954a3d6f1

MAKOMO, 2019-07-11:
Thanks for that detailed report (again)!

closed 2019-07-11 by MAKOMO