artisan-roaster-scope/artisan #378: Autosave function saving with wrong date and time

lucapinello, 2019-07-13:
## Expected Behavior
I used the Autosave option you have described (see attached screenshoot). My local time is 9:51AM Jul 13 2019. 

## Actual Behavior
But the file get saved to:
1) artisan_log_19-07-10_2013.alog 
and
2) artisan_log_19-07-10_2013.pdf

So the date is wrong and the time is missing

[image]

## Specifications
[image]
Windows 10 machine

roasterdave, 2019-07-13:
Can you tell me the steps you took?  Was this a roast in progress or a completed roast?

lucapinello, 2019-07-13:
Thanks for the quick reply! I tried with a new one and it worked as expected. I was fooled by the date format and assumed 2013 was the year. 
[image]

closed 2019-07-13 by lucapinello