artisan-roaster-scope/artisan #380: the problem ofroast properties

yunzi-jing, 2019-07-18:
hello～i am using theartisan-mac-2.0.0.dmg
but boast roast properties is available,click the button display
<class'valueerror’>:
“is not list
how should i solve this problem

MAKOMO, 2019-07-18:
Could you please provide a full error log or a screenshot of the issue. Please also attach your Artisan settings file (save using menu Help >> Save Settings). Thanks!

yunzi-jing, 2019-07-18:
Sorry, I don't know how to upload pictures here

yunzi-jing, 2019-07-18:
May I have your email address? I can send the picture to you by email

MAKOMO, 2019-07-18:
Just drag and drop it on the edit field. Same for the settings file.

yunzi-jing, 2019-07-18:
[image]

yunzi-jing, 2019-07-18:
Oh, thank you ~ this is the text displayed after the properties button in the store

yunzi-jing, 2019-07-18:
I use version 0.9.7 of the software, this function is available

MAKOMO, 2019-07-18:
Sorry, we can only support the current version which is v2.0. There is no way to fix anyting in older version. v0.9.7 is already 4 years old!! Please update to the current version and check if the issue persist. Sorry.

yunzi-jing, 2019-07-19:
The picture I showed you is the problem with the 2.0 version of the software

yunzi-jing, 2019-07-19:
[image]

MAKOMO, 2019-07-19:
Hm. Could you indicate in detail how I could reproduce this bug. I do not see a button "Display" in the "Roast Properties" dialog. Could you please also attach your Artisan settings file (again: save via menu Help >> "Save Settings" into an aset file). Thanks!

yunzi-jing, 2019-07-19:
Thank you, problem solved

yunzi-jing, 2019-07-19:
I would like to ask, does Artisan have detailed instructions for each function? There are so many things I don't understand how to use

MAKOMO, 2019-07-19:
Could you please indicate how you solved the issue? The message looked like indicating there is a bug. I would love to fix this for others too. However, I need to be able to reproduce it here and thus I need your instructions on how to do this and also your Artisan settings. Thans!

MAKOMO, 2019-07-19:
There is online documentation available from the Artisan home page https://artisan-scope.org/. There is also a link to the Artisan blog which contains a lot information and has an Overview page to quickly find the post you are interested in.

yunzi-jing, 2019-07-19:
Well, the baking curves I saved were saved in version 0.9.7. After opening the old ones with 2.0, the words I gave you in the picture will appear, which cannot be changed. After this problem occurred, I opened a new curve file saved with 2.0 version of the software, and the software returned to normal. This problem was also caused by my improper operation. Thank you for your help ~ this is from China

yunzi-jing, 2019-07-19:
Ok, wait a minute I'll do the old one

closed 2019-10-11 by MAKOMO