artisan-roaster-scope/artisan #386: Batch counter overwrite notification dialgue: clearer language

InvisibleOrdinary, 2019-08-07:
## Expected Behavior

Just a suggestion to make the alert more clear when you open a settings file with a different batch counter. The current wording isn't very clear, and somewhat misleading.

Currently, it states: "Overwrite your current batch counter 239 by 205 from the settings file?"

This is actually saying that if you choose to overwrite, the counter will now be set at 444.

Suggested change: "Do you want to update the batch counter for this setting file with the current counter? Yes will set it to 239. No keeps the setting file's current value (205).

it would also be preferable to set this in the app preferences - whether to use a perpetual batch counter, or separate for each settings file. I think I saw a similar request recently, but lost track of that one.

Or maybe this is an app preference and just don't know where to set that?

## Specifications

  - Artisan Version: 2.0.0 (a0a1fac) 
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): MacOS 10.14.6
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-08-19:
Well, the settings files hold the app preferences. Artisan is not designed to be used in production on several roasting machines from the same computer. Usually, those operations have one laptop/Artisan instance running per machine and thus the issue does not occur.

Your suggestion ""Do you want to update the batch counter for this setting file " is more confusing as it is not the batch counter of the settings file that is changed, but the counter of the currently running Artisan instance on importing that settings file.

Things have now be improved in two ways. For one, the batch counter of the last loaded settings file is automatically updated together with the counter of the running instance such that if that same setting file is loaded later again, the counter starts at the same point. Further the message on counter miss matches was slightly extended to clear indicate that the source counter is from the settings file to be imported.

Thanks for your suggestions!

closed 2019-08-19 by MAKOMO

InvisibleOrdinary, 2019-08-19:
Hmmm. I'm not sure I'm following you on this one.

Marko Luther wrote on 8/19/19 10:20 AM:
>
> Well, the settings files hold the app preferences. Artisan is not 
> designed to be used in production on several roasting machines from 
> the same computer. Usually, those operations have one laptop/Artisan 
> instance running per machine and thus the issue does not occur.
>
I do use the same computer on both my sample roaster and production 
roaster… so while that may not be an intention when designing, I'm going 
to guess that I am not the only one who moves the computer between a 
couple different locations, not just different different settings for 
different operators…
>
> Your suggestion ""Do you want to update the batch counter for this 
> setting file " is more confusing as it is not the batch counter of the 
> settings file that is changed, but the counter of the currently 
> running Artisan instance on importing that settings file.
>
What ever wording is most clear, then cool.
> Things have now be improved in two ways. For one, the batch counter of
> the last loaded settings file is automatically updated together with
> the counter of the running instance such that if that [...]

This isn't clear to me… so if I have an Artisan instance with a batch 
counter currently at 500, and open a different setting file that its 
last batch was 350, then it will automatically revert to 350, or start 
counting at 500? If this is 500, cool (for how I use it). If it starts 
counting at 350 again, that wold be exactly the opposite from how I use 
this - one significant, and potentially legal reason why I would need to 
keep all batches sequential (or at least unique), is that I need to 
retain batch numbers / logs for food safety regulations. If I end up 
with two or more batches with the same batch number, and I need to go 
back to them for any reason, then I have little way of telling which 
batch was bad… insaddition to the food safety requirements, in terms of 
tracking for safety, it also opens up a liability.

Thanks

Dave

-- 
Dave Ewald
Deft Coffee Roasters  | 503.702.1932 | amazing coffees delivered 
<http://deftcoffee.com>
<http://deftcoffee.com>

MAKOMO, 2019-08-21:
Ok, Dave. If your batch counter is at 500 and you load a settings file with the counter at 350, you will be asked by a popup "Overwrite your current batch counter 500 by 350 from the settings file to be imported?" and you can either abort the loading of the settings file by pressing "Cancel", or say "No" and the settings file will be imported but your counter will stay at 500, or you say "Yes" and the settings file will be imported and the counter will be set to 350 from that setting file.

So for your case you would just always answer this popup with "No" to keep your batch counter untouched. 

To make this even simpler for your case, I added a flag to the batch counter dialog (yes, another configuration option:( to "Never overwrite counter"). If you tick this flag, by default off, this popup dialog on loading a settings file with a different batch counter state will always be answered automatically with "No" and thus not shown. I assume this is the behaviour you intend for your case.

A beta builds with this extension are available here:
https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous