artisan-roaster-scope/artisan #390: Can't access additional Phidget device

tonescout, 2019-08-16:
## Expected Behavior

adding additional device to read a thermocouple attached I had hoped to read a 3rd temperature probe in addition to the two RTD devices that are mapped to BT and ET, and working well.

## Actual Behavior
I can see this in the Phidget control panel, but if I try to access it through the Vint Hub, or as the ports in the Phidget TMP1101 ports it does not read.

## Steps to Reproduce the Problem

  1.  tried to access via Hub
  2. tried to access via TMP1101 
  3. tried different hub ports on VINT

## Specifications

  - Artisan Version: 2.0 
  - Artisan Build (number in brackets shown in the about box): (a0a1fac)
  - Platform (Mac/Windows/Linux + OS version): OS 10.14.6
  - Connected devices or roasting machine:

Phidget Vint Hub 
Phidget TMP 1200 (2 connected to port 0, and port 1)
Phidget i-Snail (for roaster voltage supplied on port 4)
Phidget TMP 1101 (port 5 not working in VIONT hub to read Port 0 in TMP 1101)
Phidget 1125 ambient temp sensor only currently used for room Temp (vint port 2)

USB to MAcBook
[artisan-settings-RP.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/3510423/artisan-settings-RP.aset.zip)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-08-16:
The TMP1101 on port 5 cannot be accessed as the extra device of type "Phidget HUB IO 45" used to read the i-Snail "mounts" both ports the analog input port 4, but also the analog input port 5. Phidget HUB IO ports are always mounted and accessed in pairs. Therefore the Phidgets driver is not able to access the TMP1101 connected to that port 5. 

If you connect the Phidget iSnail to that port 3 (instead of port 4) it should be read as the second channel of that extra device of type "Phidget HUB0000 IO 23" and you could connect the TMP1101 to port 4, leaving the port 5  for future use.

The next version of Artisan will feature a "Phidget HUB IO 0" device that allows to mount just one HUB channel to allow the use of all HUB ports even if an uneven number of analog or digital inputs are used. The current [pre-release builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) already contain this device type.

closed 2019-08-16 by MAKOMO

tonescout, 2019-08-17:
this is great - it works.

I wonder if there is anyway to see an ROR of the 3rd probe to detect changes more easily?

Thanks!!

R

MAKOMO, 2019-08-17:
The RoR deltas are only computed for ET and BT. You can at add a RoR for any other curve using a Virtual Extra Device and a corresponding symbolic formula. 

https://artisan-roasterscope.blogspot.com/2015/10/signals-symbolic-assignments-and-plotter.html

Those will however not benefit from the specific smoothing support for the ET/BT deltas.

M