artisan-roaster-scope/artisan #393: beta (583c2d8) LCD issue

InvisibleOrdinary, 2019-09-04:
## Expected Behavior
show same temp as side LCD

## Actual Behavior
If the window is made narrow enough, it won't resize to show the entire temp, and only show last 3 digits of the probe (eg, 235.5 become 35.5)
This is resolved if you drag the window wide enough.

## Steps to Reproduce the Problem

  1. open LCD on 2.0.1(583c2d8)
  2. resize LC window to see how it shifts between narrow and wide
  3. 

## Specifications

  - Artisan Version: 2.0.1(583c2d8) (pre-release beta)

  - Platform (Mac/Windows/Linux + OS version): Mac OX 10.14.6
  - Connected devices or roasting machine: phidgets & Proaster
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-09-04:
Dear Dave, thanks for the quick review. I assume you are talking about the large "Main LCDs", right?  Could you please attach an image that demonstrates this. I cannot reproduce this on my Mac.

MAKOMO, 2019-09-05:
Confirmed and fixed by correcting the LCDs digit count in 3b7363826d45c65e892c7d255cba1375fc9c383d

closed 2019-09-05 by MAKOMO