artisan-roaster-scope/artisan #394: beta (583c2d8) LCD issue #2

InvisibleOrdinary, 2019-09-04:
## Expected Behavior
Extra LCD show the extra LCDs

## Actual Behavior

LCD is blank

## Steps to Reproduce the Problem

  1.  Artisan beta (583c2d8) LCD issue
  2. 
  3. 

## Specifications

  - Artisan Version:beta (583c2d8) 

  - Platform (Mac/Windows/Linux + OS version): 10.14.6 (in dark mode, if that matters)
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-09-04:
Hm. If no extra LCDs are define (or visible) the large Extra LCD window will stay blank. Is this the situation? Could you please attach your .aset settings file to allow me to check?

MAKOMO, 2019-09-06:
Got it now. Visibility of extra large LCDs was respecting curve visibility instead of LCD visibility settings. Fixed that in commit 15d4e0c35f035e8929b6bfb06572dd276c2845e5.

closed 2019-09-06 by MAKOMO