artisan-roaster-scope/artisan #395: saving glitch? saved with incorrect date

InvisibleOrdinary, 2019-09-04:
## Expected Behavior

open a past .alog and making changes… when saving, it should be saving the same file, same file name…

## Actual Behavior

This one is convoluted, not the least of which is because it took me a bit to realize what was wrong.

I had had several roasts that did not retain the roast description data* (yes, 'DELETE ROAST properties on RESET' was unchecked) and after roasting, went back to change the information to be correct. This included beans (linked to artisan+ and weight. I open, correct, and save over the file.. on one file, it arbitrarily changed the batch number from 263 to 163, and the date from 9/3 @ 20:33 to 9/4 @ 20:48 (which is actually in the future, or maybe German time). 

There'a a related error/issue in that it's not possible to edit the batch number, so now, the batch number says '163 (3)', and is no longer linked to the roast in artisan+, plus it seems that the new roast (163 (3) is in artisan+ with incorrect info.

I'll point out again that it took a minute to realize what wasn't working, so there were a few things I did that may or may not have played into this. They were: 
 - opened artisan+ 
 - opened & saved the file in the most recent stable version
 - opened other files and re-saved
 - ended up renaming the .alog file, however the batch # is still incorrect

* That's a separate issue, and one that seemed to clear itself up, so not sure how to submit it when I couldn't reproduce it. 

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.0.1 (583c2d8) 
  - Platform (Mac/Windows/Linux + OS version): Mac 10.14.6
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

[Archive.zip](https://github.com/artisan-roaster-scope/artisan/files/3573308/Archive.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-09-04:
Dear Dave, this is interesting despite that it will get hard to find out what happend exactly as this report is somewhat confusing. Just for your understanding: each alog profile gets a unique ID assigned that is used to match the entries on the plus platform with the local file. If you just save an alog profile in Artisan after applying some changes, this UUID is preserved. This holds also if you save a profile under a different name via "Save As".  If, however, you use the new "Save a Copy As.." the newly saved file gets a different unique ID assigned and is not synced to the platform. Not sure if this is what you did.

Anyhow, batch numbers should not be changed randomly nor dates. Also roast properties should not get lost.

I will look into your data and files later to find out whats the current state of things on your plus account.

Thanks for this report!

MAKOMO, 2019-09-04:
Ah, Dave. What would help is to learn your usual procedure on doing a roast and adding the roast properties incl. the sync with the plus platform. Eg. are you first adding the coffee/batch-size before doing the roast, or are you adding those later (via the client or the platform)....

Thanks,
Marko

MAKOMO, 2019-11-06:
Closing this as no additional information was received to track this down

closed 2019-11-06 by MAKOMO