artisan-roaster-scope/artisan #396: How to avoid automatic reset with Autosave on?

lucapinello, 2019-09-29:
I am using the autosave function to automatically generate a PDF report. This works great, however, the view is automatically reset when I hit stop. How can I keep the curves until I hit again START?

Without the autosave function, I have the option to keep the curves if needed.

  - Artisan Version:2.0
  - Platform (Mac/Windows/Linux + OS version): Windows

 
Thanks!

MAKOMO, 2019-09-30:
Dear Luca, we cannot reproduce this on either Mac OS X nor Windows running Artisan v2.0. Could you please attach your Artisan settings here to let us see if there is something specific in your configuration that makes this effect? Save your settings via menu Help >> Save Settings into an .aset file and zip it before you just drag and drop it here into an answer.

Thanks!

lucapinello, 2019-09-30:
Hi @MAKOMO absolutely! Thanks a lot for looking into this.
[artisan-settings.zip](https://github.com/artisan-roaster-scope/artisan/files/3671136/artisan-settings.zip)

MAKOMO, 2019-09-30:
Well, in your settings you instructed Artisan to turn immediately ON again after the OFF action by ticking "Keep ON" under Config >> Sampling Interval. The ON action has to do a reset to get rid of the data of the previous roast. So just untick that one and you will see your profile after OFF as expected. Note that this has nothing to do with the Autosave feature. Just if the Autosave feature is not set, the RESET (triggered by the ON, triggered by having "Keep ON" ticked) will ask for a path to save the previous data, while with Autosave on, that data got already saved.

Btw. defining more than 180 alarms to adjust Fan and Burner sliders does not make much sense to me. Take a look at the replay feature with allows to replay such events from a background profile (which you can draw using the designer and add the slider events via the event table manually).

closed 2019-09-30 by MAKOMO

lucapinello, 2019-10-01:
Thank you so much for the quick reply! I totally missed this. 

Also thanks for the suggestion on the designer, this is something that I need to explore for sure! My past solution was to write a program to draw these two curves and generate an alarm file for artisan: https://github.com/lucapinello/SR700_Artisan_Profile_Builder.