artisan-roaster-scope/artisan #398: 'off' at end of a series of batches should auto save last roast when autosave is enabled

InvisibleOrdinary, 2019-10-03:
## Expected Behavior

when using autosave and cmd-n to run a sequence of roasts, the last roast should be autosaved when hitting 'off'

## Actual Behavior

last roast isn't autosaved

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.0.1 (95be8ff)
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14.6 (18G95)
  - Connected devices or roasting machine: phidgets 1048
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-10-04:
I cannot duplicate this. Only "roasts" that do not contain at least 3 measurements are not stored on OFF by the autosave mechanism. How that last roast started does not play a role (or should not). Could you give a hint on a minimal setup and process to reproduce this issue?

InvisibleOrdinary, 2019-12-05:
Closing. This seems to have resolved since I submitted this issue. Will submit a new ticket if I see it come up again.

closed 2019-12-05 by InvisibleOrdinary