artisan-roaster-scope/artisan #400: switching setting profile, extras viewed isn't visible until brofile bg opened a second time

InvisibleOrdinary, 2019-10-09:
## Expected Behavior

extras available would match the extras in the active settings profile

## Actual Behavior

1st time opening profile shows previous profile extra devices
2nd time shows correct / current extras devices

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.0.1 (95be8ff) 

  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14.6 (18G95)
  - Connected devices or roasting machine: phidgets 1048
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-10-09:
Could you please give some more information on how to reproduce this? It would be good to have a sequence starting from a factory reset (or a given .aset settings file) with two .alog profiles that would reproduce this effect. Again I cannot reproduce this. Thanks!

InvisibleOrdinary, 2019-12-05:
I'm going to close this. I haven't seen it come up since a couple revs ago, and we are on a new release. If it comes up again, I'll submit a new ticket.

closed 2019-12-05 by InvisibleOrdinary